theori-io/nrsc5 #211: OSX: cmake Error 2

liftereyes, 2019-12-25:
Hello,

I am using the command brew install --HEAD https://raw.githubusercontent.com/theori-io/nrsc5/master/nrsc5.rb to install this on Mac. However, the install gets to about 91% and fails.

This is the output I am getting:

==> Cloning https://github.com/theori-io/nrsc5.git
Updating /Users/timothygreenli/Library/Caches/Homebrew/nrsc5--git
==> Checking out branch master
Already on 'master'
Your branch is up to date with 'origin/master'.
HEAD is now at 43c8b21 Merge pull request #208 from argilo/fix-wav-output-on-windows
==> cmake .. -DCMAKE_C_FLAGS_RELEASE=-DNDEBUG -DCMAKE_CXX_FLAGS_RELEASE=-DNDEBUG
==> make install
Last 15 lines from /Users/user/Library/Logs/Homebrew/nrsc5/02.make:
Scanning dependencies of target nrsc5_static
/Library/Developer/CommandLineTools/usr/bin/make -f src/CMakeFiles/nrsc5.dir/build.make src/CMakeFiles/nrsc5.dir/build
/Library/Developer/CommandLineTools/usr/bin/make -f src/CMakeFiles/nrsc5_static.dir/build.make src/CMakeFiles/nrsc5_static.dir/build
make[2]: *** No rule to make target `rtlsdr-prefix/lib/librtlsdr_static.a', needed by `src/libnrsc5.dylib'.  Stop.
make[1]: *** [src/CMakeFiles/nrsc5.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 91%] Linking C static library libnrsc5_static.a
cd /tmp/nrsc5-20191224-39836-102nfx1/build/src && /usr/local/Cellar/cmake/3.16.2/bin/cmake -P CMakeFiles/nrsc5_static.dir/cmake_clean_target.cmake
cd /tmp/nrsc5-20191224-39836-102nfx1/build/src && /usr/local/Cellar/cmake/3.16.2/bin/cmake -E cmake_link_script CMakeFiles/nrsc5_static.dir/link.txt --verbose=1
/usr/bin/ar qc libnrsc5_static.a  CMakeFiles/nrsc5_object.dir/acquire.c.o CMakeFiles/nrsc5_object.dir/decode.c.o CMakeFiles/nrsc5_object.dir/frame.c.o CMakeFiles/nrsc5_object.dir/input.c.o CMakeFiles/nrsc5_object.dir/nrsc5.c.o CMakeFiles/nrsc5_object.dir/output.c.o CMakeFiles/nrsc5_object.dir/pids.c.o CMakeFiles/nrsc5_object.dir/sync.c.o CMakeFiles/nrsc5_object.dir/firdecim_q15.c.o CMakeFiles/nrsc5_object.dir/conv_dec.c.o CMakeFiles/nrsc5_object.dir/rs_init.c.o CMakeFiles/nrsc5_object.dir/rs_decode.c.o CMakeFiles/nrsc5_object.dir/log.c.o CMakeFiles/nrsc5_object.dir/unicode.c.o CMakeFiles/nrsc5_object.dir/strndup.c.o
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: libnrsc5_static.a(strndup.c.o) has no symbols
/usr/bin/ranlib libnrsc5_static.a
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: libnrsc5_static.a(strndup.c.o) has no symbols
[ 91%] Built target nrsc5_static
make: *** [all] Error 2

Do not report this issue to Homebrew/brew or Homebrew/core!

argilo, 2020-01-25:
Which version of MacOS are you using? I used Travis to test on MacOS 10.13 and 10.14, and 
`brew install --HEAD https://raw.githubusercontent.com/theori-io/nrsc5/master/nrsc5.rb` worked fine on both.

argilo, 2020-05-18:
I also tested on MacOS 10.15 and didn't encounter any problems. I'll close this issue for now, since I'm unable to reproduce the problem.

closed 2020-05-18 by argilo