theori-io/nrsc5 #212: Awful sounding output?

Qwertylex, 2020-01-09:
It almost sounds like timestretched / wrong samplerate + echoic
[fail.tar.gz](https://github.com/theori-io/nrsc5/files/4043101/fail.tar.gz)

I'll upload IQ samples if needed

argilo, 2020-01-09:
I would guess that's a problem with the audio codec. Do you have a hardware HD Radio receiver to compare against, to make sure it's not the broadcaster that's at fault?

It would be helpful if you could record an IQ file (using the `-w` command line option) so we can investigate further.

Qwertylex, 2020-01-10:
25 seconds of IQ is around 76MB gzipped, github allows up to 10MB for attaching to issues, should I use `split` and upload the resulting 8 files?
Or do you recommend another place to upload as a single file?

awesie, 2020-01-10:
Would send.firefox.com work?

Qwertylex, 2020-01-11:
https://send.firefox.com/download/b2a2d3ab3481779f/#Lt9cPjGqKrU0oJh7zHVUCQ

The SHA-512 sum of the un-gzipped tar file should match:
c7af9c4b1bf6e547a613c82a638e9c416dc2539f3e720314aa00d34981e18225722fdf2fef67b82553d7c3caa6b78441e5d9f21552676e9d5439d9f751d1f599

the tar includes the HDC, IQ and WAV, as well as debug output from nrsc5

awesie, 2020-01-11:
Thanks for uploading.

I think it sounds correctly for me running: `nrsc5 -r issue212/iq/2020-01-10T14-29-31-0500.iq 0`. @argilo what do you think?

pclov3r, 2020-01-11:
It seems to sounds correctly here as well so I suspect something else is going on.

Perhaps some buffer issue someplace?

awesie, 2020-01-11:
I immediately suspected it could be related to the compiler since I've seen issues crop up in other projects with recent compilers.

Testing on Linux, gcc 7.4.0 sounds correct and gcc 9.2.1 sounds incorrect (e.g. the same as the wav posted above). The AAC decoder, FAAD2, is not the cleanest code, so I suspect some recent compiler optimization is causing a problem.

argilo, 2020-01-19:
We should probably leave this open until the root cause is known.

argilo, 2020-01-19:
A couple additional notes:

* `-DUSE_SSE=ON` is needed to replicate the issue
* adding `#pragma GCC optimize ("O2")` to the top of `libfaad/mdct.c` makes the problem go away, which suggests the problem lies in that file

argilo, 2020-01-19:
Compiling with `-fopt-info-vec-optimized` shows the following optimizations for that file:

```
/home/argilo/nrsc5/build/faad2-prefix/src/faad2_external/libfaad/mdct.c:176:5: optimized: loop vectorized using 16 byte vectors
/home/argilo/nrsc5/build/faad2-prefix/src/faad2_external/libfaad/mdct.c:176:5: optimized:  loop versioned for vectorization because of possible aliasing
/home/argilo/nrsc5/build/faad2-prefix/src/faad2_external/libfaad/mdct.c:158:5: optimized: loop vectorized using 16 byte vectors
/home/argilo/nrsc5/build/faad2-prefix/src/faad2_external/libfaad/mdct.c:158:5: optimized:  loop versioned for vectorization because of possible aliasing
```

awesie, 2020-01-19:
Thanks for narrowing that down. I played with it a bit, and the culprit seems to be the pre-IFFT complex multiplication loop in faad_imdct, though it looks fine to me.

I slightly refactored the code, and I think changing the uint16_t loop index variable to unsigned int "fixed" the problem. Here is a patch if you want to investigate more: https://gist.github.com/awesie/5142bbcf2953d02b3558b59326c97086.

argilo, 2020-01-24:
I also came to the same conclusion that it's the pri-IFFT loop, and I didn't spot anything wrong with the C code there. Even just casting the `N2 - 1 - 2*k` index to `unsigned int` "fixes" the problem, but only because the compiler doesn't decide to vectorize the loop in that case. I suspect a compiler bug.

awesie, 2020-01-25:
I turned it into a single file test case: https://gist.github.com/awesie/7c6f43c3117ea82cbd832551ee384100. I'll test with latest gcc snapshot and file a bug report if the issue persists.

awesie, 2020-01-25:
Test case passes with `gcc version 10.0.0 20191208 (experimental) (GCC)`. Closing as "not our bug". I'll keep the optimization level at `-O2` so we can avoid this issue.

closed 2020-01-25 by awesie

argilo, 2020-01-26:
Bug references:
https://gcc.gnu.org/bugzilla/show_bug.cgi?id=92420
https://bugs.launchpad.net/ubuntu/+source/gcc-9/+bug/1860876

PR #213 (2020-01-11): Compile FAAD2 with -O2 instead of -O3.
GCC 9 with -O3, specifically with auto vectorization enabled, causes audio
corruption. I do not know how to find the root cause, so just disable the
extra optimizations for now.

Fixes #212.
commit aebb0179 (2020-01-11): Compile FAAD2 with -O2 instead of -O3.

GCC 9 with -O3, specifically with auto vectorization enabled, causes audio
corruption. I do not know how to find the root cause, so just disable the
extra optimizations for now.

Fixes #212.