theori-io/nrsc5 #216: Newbie No audio on Pinebook Pro able to save raw IQ data to file

QUANTA901, 2020-02-14:
nrsc5 81 1
Gives:
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.front // Error?
Found Elonics E4000 tuner // using nooelec elonics e4000
Exact sample rate is: 1488375.071248 Hz
08:42:09 DEBUG nrsc5.c:57: Gain: -1.0 dB, CNR: -0.0 dB
08:42:09 DEBUG nrsc5.c:57: Gain: 1.5 dB, CNR: 0.1 dB
08:42:09 DEBUG nrsc5.c:57: Gain: 4.0 dB, CNR: 0.2 dB
08:42:09 DEBUG nrsc5.c:57: Gain: 6.5 dB, CNR: 0.2 dB
08:42:09 DEBUG nrsc5.c:57: Gain: 9.0 dB, CNR: 0.5 dB
08:42:09 DEBUG nrsc5.c:57: Gain: 11.5 dB, CNR: -0.2 dB
08:42:09 DEBUG nrsc5.c:57: Gain: 14.0 dB, CNR: 0.3 dB
08:42:09 DEBUG nrsc5.c:57: Gain: 16.5 dB, CNR: 0.3 dB
08:42:09 DEBUG nrsc5.c:57: Gain: 19.0 dB, CNR: -0.6 dB
08:42:09 DEBUG nrsc5.c:57: Gain: 21.5 dB, CNR: -0.5 dB
08:42:09 DEBUG nrsc5.c:57: Gain: 24.0 dB, CNR: -0.4 dB
08:42:09 DEBUG nrsc5.c:57: Gain: 29.0 dB, CNR: -1.4 dB
08:42:10 DEBUG nrsc5.c:57: Gain: 34.0 dB, CNR: -1.3 dB
08:42:10 DEBUG nrsc5.c:57: Gain: 42.0 dB, CNR: -1.6 dB
08:42:10 DEBUG nrsc5.c:66: Best gain: 9.0 dB, CNR: 0.5 dB

awesie, 2020-02-14:
We use libao to play audio, which defaults to using ALSA on Linux. It appears from the error message in the output that either libao or ALSA is failing. My suggestion is to investigate how to fix your local ALSA and/or libao configuration (e.g. asound.conf, libao.conf).

awesie, 2020-02-14:
I will also add that the CNR numbers in your output are very low, so it might just be the case that you are not actually decoding anything. In that case, the ALSA warning may not be important.

argilo, 2020-02-14:
You have specified a frequency of 81 MHz, which is outside the FM band (88-108 MHz). That would explain why you're not receiving anything. I would suggest double checking the frequencies of your local HD Radio stations.

argilo, 2020-02-14:
I'll close this for now, since it doesn't appear to be a problem with nrsc5.

closed 2020-02-14 by argilo

QUANTA901, 2020-02-14:
88.1 1

QUANTA901, 2020-02-14:
ok

QUANTA901, 2020-02-14:
#libao.conf
default_driver=alsa
verbose

nrsc5 89.5 0 
ERROR: Failed to load plugin /usr/lib/ao/plugins-4/libesd.so => dlopen() failed
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.front
ao_alsa WARNING: Unable to open surround playback.  Trying default device...

Found Elonics E4000 tuner
Exact sample rate is: 1488375.071248 Hz
19:20:18 DEBUG nrsc5.c:57: Gain: -1.0 dB, CNR: -0.0 dB
19:20:18 DEBUG nrsc5.c:57: Gain: 1.5 dB, CNR: 0.1 dB
19:20:18 DEBUG nrsc5.c:57: Gain: 4.0 dB, CNR: 0.0 dB
19:20:18 DEBUG nrsc5.c:57: Gain: 6.5 dB, CNR: -0.2 dB
19:20:19 DEBUG nrsc5.c:57: Gain: 9.0 dB, CNR: -0.2 dB
19:20:19 DEBUG nrsc5.c:57: Gain: 11.5 dB, CNR: -0.2 dB
19:20:19 DEBUG nrsc5.c:57: Gain: 14.0 dB, CNR: 0.3 dB
19:20:19 DEBUG nrsc5.c:57: Gain: 16.5 dB, CNR: -0.2 dB
19:20:19 DEBUG nrsc5.c:57: Gain: 19.0 dB, CNR: -0.1 dB
19:20:19 DEBUG nrsc5.c:57: Gain: 21.5 dB, CNR: -0.6 dB
19:20:19 DEBUG nrsc5.c:57: Gain: 24.0 dB, CNR: -0.5 dB
19:20:19 DEBUG nrsc5.c:57: Gain: 29.0 dB, CNR: 0.3 dB
19:20:19 DEBUG nrsc5.c:57: Gain: 34.0 dB, CNR: 0.8 dB
19:20:19 DEBUG nrsc5.c:57: Gain: 42.0 dB, CNR: 6.2 dB
19:20:19 DEBUG nrsc5.c:66: Best gain: 42.0 dB, CNR: 6.2 dB

QUANTA901, 2020-02-14:
#asound.conf
pcm.!default {
    type             plug
    slave.pcm       "rock_audio_pot"
}

pcm.rock_audio_pot {
    type            softvol
    slave {
        pcm         "hw:1"
    }
    control {
        name        "Master"
        card        1
    }
}

QUANTA901, 2020-02-14:
now its only giving ERROR: Failed to load plugin /usr/lib/ao/plugins-4/libesd.so => dlopen() failed

QUANTA901, 2020-02-14:
weird

QUANTA901, 2020-02-14:
How can I channel raw IQ data to a program and play it? like nrsc5 89.5 0 -w - | #code