theori-io/nrsc5 #218: Audio died midstream. Station details and bit rate work but no sound

Prerunning, 2020-04-08:
Tried deleting everything and repeating the install but problem persist.
Sample audio plays correctly. I start it from msys2 and tune to 0 or 1 but same problem. Station details update but no sound
[image]

argilo, 2020-04-08:
Could you make an I/Q recording using the `-w` option and share the resulting file? Thanks.

Prerunning, 2020-04-12:
> Could you make an I/Q recording using the `-w` option and share the resulting file? Thanks.

Sorry i would have replied sooner but i didnt see there was a reply. Shows 1 comment. But here is the -w you requested.
https://drive.google.com/open?id=1VTavmBlmmsEACXfkTUA08j3sMJt4D-xt

argilo, 2020-04-20:
@Prerunning Please keep in mind that this project is maintained by volunteers in our spare time. We'll have a look at this soon, but I can't make any promises when exactly that will be.

argilo, 2020-04-21:
The problem here lies with KJJJ-FM, not nrsc5. Using a USRP B200, I replayed your recorded signal and both of my hardware HD Radio receivers played silence once they locked onto the HD signal. So it would seem the station really is broadcasting silence.

closed 2020-04-21 by argilo

Prerunning, 2020-04-21:
Thank you for verifying that, I would have been troubleshooting forever.