theori-io/nrsc5 #219: Windows 10 MSYS2 install issues

zaxk2137, 2020-05-15:
**I'm very inexperienced with both Linux and Git**
Just thought I'd share some issues I had and how I solved them:

My setup is a Windows 10 PC, I'm using MSYS2 [https://www.msys2.org/]
I had to run "pacman -S make" and "pacman -S cmake" additionally to the other commands as I kept getting errors that make / cmake commands were not valid.
(Easy enough for even me to figure out with some googling)

My biggest issue, however, was this:
when running "nrsc5/support/msys2-build", MSYS would get caught up at "exporting_libao" and after several hours are troubleshooting I found out it's because it's trying to reach this URL:
https://git.xiph.org/libao.git
Which happens to not be valid (I'm assuming it was when this was compiled)
So after several more hours of dicking around, I was able to get it to work by:
Opening "CMakeLists.txt" in ~
Changing "https://git.xiph.org/libao.git" on line 187 to:
"https://github.com/xiph/libao.git"
I believe all I did was change a now invalid address for cloning libao to one the is currently working.
I hope this helps someone not spend hours of headbanging like I did.

cjbarker, 2020-05-18:
I ran across the same issue today when installing on Windoze 10.  PR submitted that fixes issues of unresolvable domain name timeout.

https://github.com/theori-io/nrsc5/pull/220

argilo, 2020-05-18:
Closing, since this was fixed in #220.

closed 2020-05-18 by argilo