theori-io/nrsc5 #221: Single iBOC sideband decoding?

pclov3r, 2020-06-14:
So it appears that if a sideband is completely blocked or has horrible multipath NRSC5 will not decode any audio. Other hardware radios seem to be able to lock and decode with only a a good single iBOC sideband present. 

Not sure what the NRSC-5 spec says about this or if it should or shouldn't be supported.

Thanks!

argilo, 2020-06-14:
One reason this doesn't currently work is that the synchronization code requires two particular reference subcarriers to be present, one in the lower sideband and one in the upper sideband:

https://github.com/theori-io/nrsc5/blob/17181bb604ede55e5be6fba372b4865cb1e0a5aa/src/sync.c#L226-L237

If one sideband is gone (or even if narrow-band noise knocks out one of those two specific subcarriers) then synchronization will fail.

Synchronization should be done in a more flexible way.

argilo, 2020-06-20:
@pclov3r I've fixed this in #223.

> I've some some people claim that with a station that only has 2 HD streams both sidebands contain duplicate information. For HD3 and 4 stations both need to be present. Not sure how true this is.

The difference is that the P1 logical channel (which typically contains the HD1 & HD2 subchannels) uses a rate 2/5 error-correcting code (which can tolerate up to 60% of the data being missing), whereas the P3 logical channel (which typically contains the HD3 subchannel) uses a rate 1/2 error-correcting code (which can only tolerate 50% of the data being missing). A missing sideband knocks out 50% of the data, leaving the P3 logical channel on the fringe of decodability. Neither of my hardware HD Radio receivers can decode it, but nrsc5 with #223 can (albeit with the occasional audio packet lost due to a CRC error).

pclov3r, 2020-06-20:
Thanks for the patch @argilo I'll give this a try later today if I can. 

Also, This patch will probably give a higher chance at decoding and for people using this for DXing conditions.

PR #223 (2020-06-21): Consider all reference subcarriers when detecting CFO
Fixes #221.

As noted in https://github.com/theori-io/nrsc5/issues/221#issuecomment-643716708, CFO synchronization uses two specific reference subcarriers. If either of those subcarriers is missing (due to interference or multipath), then the entire synchronization process fails.

To fix this, I've changed the CFO synchronization routine to take into account all 22 reference subcarriers in the Primary Main (PM) sidebands. If at least three of them are found to have the same offset, then we assume that we've found the correct synchronization [... cut]
closed 2020-06-21 by argilo