theori-io/nrsc5 #233: Add feature to repeat playback when reading I/Q files

pclov3r, 2020-08-06:
I figure this wouldn't be difficult to implement but it would be nice to be able to repeat the playback when reading I/Q files instead of having to start over again.

This would be useful for switching between the different streams during playback.

pclov3r, 2022-06-27:
Alternatively, I'd be happy when using the -R IQ input option if NRSC5 closed at the end of the I/Q file and there is no more audio playback. 

You have to manually close it anyways if you want to read another I/Q file.

argilo, 2025-02-21:
Perhaps we could close this off. I suspect repeating an I/Q file is not a common use-case, and there are other ways of accomplishing it (for instance, running nrsc5 in a loop, or looping I/Q input with something like `while true; do cat my_file.cu8; done | nrsc5 -r - 0`).

The bug where nrsc5 fails to exit after playing an I/Q file now has its own separate issue: #314.

closed 2025-02-21 by argilo