theori-io/nrsc5 #237: Add option for frequency offset to support upconverters.

pclov3r, 2020-08-13:
Sorry about the 3rd issue in a row but i figure that these would be better as as separate issues rather then one issue.

With that said a option to have a frequency offset to allow use of upconverters would be nice. Use of a upconverter tends to yield better performance then using direct sampling mode on the RTL-SDR for AM broadcast usage. 

Alternatively, This isn't strictly required as you could just apply the offset + the frequency in question and use the input frequency for NRSC-5.

I'll leave it up to you @argilo if you think the extra work here is unnecessary and if you believe so feel free to close the issue.

argilo, 2020-08-13:
> Sorry about the 3rd issue in a row

No problem. Keeping each bug or feature request in a separate issue is the way to go.

argilo, 2025-05-07:
I could see this being useful in a GUI application, but for nrsc5's built-in CLI applications I'm not sure adding an extra frequency offset input would add a lot of value. You could always ask the shell to do the arithmetic for you. For instance, with a 125 MHz upconverter, you could tune 1180 AM using:

```
nrsc5 --am $((125000000 + 1000 * 1180)) 0
```

closed 2025-05-07 by argilo