theori-io/nrsc5 #240: Implement AM all-digital mode (MA3)

argilo, 2020-08-13:
The AM receiver only supports the hybrid mode (MA1) currently. At least one station (WWFD in Frederick, Maryland) is using the all-digital mode (MA3), so it would be nice to be able to receive it. It would be very helpful to have an I/Q recording of this station.

argilo, 2020-08-29:
With some help from N3JLY who has a KiwiSDR (http://sdr.hfunderpants.com:8074/) near WWFD, I was able to make a couple recordings:

1. https://drive.google.com/file/d/1acEfDG5J1BO3sgoSiIXaxLgO6Zc9-YZz/view?usp=sharing
2. https://drive.google.com/file/d/1_yH5qODR5-zPOKVtMgHRDjDRZkuNimtE/view?usp=sharing

The following GNU Radio flow graph can be used to convert these I/Q WAV files to `.cu8` files at a sample rate of 1488375 for use with nrsc5:

https://drive.google.com/file/d/1U3hg5LU8Bihs33-J980KkpzRmT6dbJ4Z/view?usp=sharing

The flow graph also has a disabled section which can be used to transmit the signals using a HackRF. I used this to verify that the recordings were good by playing them back to my Sangean HDT-20 receiver.

cbmira01, 2021-10-01:
Looks like WWMG in Egypt Lake FL, and WFAS in White Plains NY are operating all-digital MA3 also:
https://en.wikipedia.org/wiki/WMGG
https://en.wikipedia.org/wiki/WFAS_(AM)

argilo, 2024-05-05:
MA3 is implemented in #344.

closed 2024-05-06 by argilo

cbmira01, 2024-05-25:
Clayton,

Thanks for your work on this all-digital AM demonstration. I hope it helps
raise the
awareness of MA3 as a way to extend the usefulness of the MF broadcast band.

- Calvin Miracle, Louisville KY

ferrellsl, 2024-05-25:
This is a bit off-topic, but I heard a rumor that iHeart had stopped all digital AM broadcasts. Is this true or just a nasty rumor?

PR #344 (2024-05-06): Add MA3 (all digital) service mode
Fixes #240.

Here I have implemented support for MA3, the all-digital AM service mode. I tested that it is able to receive WWFD, using a recording made with W3HFU's KiwiSDR. As with the MA1 hybrid mode, only the "core" audio stream is decoded, and the "enhanced" stream is discarded. (Improvements to the HDC decoder would be needed to decode the enhanced stream. See #245 for a previous attempt at that.) I was able to decode WWFD's station logo:

[image]

While working on this, I found a bug in gr-nrsc5's header spread calculations, which I fixed in https://git [... cut]