theori-io/nrsc5 #241: Improve AM receive quality

argilo, 2020-08-13:
The AM receiver has some deficiencies:

* The code that separates the analog audio from the secondary and tertiary subcarriers (which should be 90 degrees out of phase from the analog audio) is ad hoc. It doesn't work very well, and has a magic number that should be removed: https://github.com/theori-io/nrsc5/blob/84e9257df915dbf47b4897d6a181a61e887861a3/src/acquire.c#L230-L231
* The Viterbi decoder uses hard decisions. It would work better if it took the MER of each subcarrier (or group of subcarriers) into account to make soft decisions.

I/Q recordings of various stations in various conditions would be helpful.

pclov3r, 2020-08-25:
I wonder how much better the samples I've posted would work with a soft decision viterbi decoder? Might make decoding a bit easier and more reliable. 

With that said I figure it's a bit of work to implement.

awesie, 2020-08-26:
I think the first step is to refactor the constellation decoder -> deinterleaver -> viterbi decoder path. With AM, this path current represents each bit as one literal bit. Instead, each bit should be stored as a int8 and the constellation decoder should calculate how far the constellation points are from the ideal. The viterbi decoder should already be able to handle these soft bits and do the right thing.

argilo, 2020-08-26:
Correct, switching to soft decisions shouldn't be too hard.

I suspect that poor phase synchronization is causing bigger problems than the lack of soft-decision decoding though. The amplitude of the analog AM signal is much larger than the amplitude of the secondary & tertiary digital subcarriers that overlap with it, so interference becomes a big problem if synchronization is off by only a few degrees.

argilo, 2020-08-26:
Another deficiency in the OFDM synchronization (which I forgot to copy over from https://github.com/theori-io/nrsc5/issues/33#issuecomment-667778808) is that there's no averaging to cancel out the noise in the training/pilot symbols. This results in extra noise being added to the data symbols.

awesie, 2020-08-29:
I've been trying to understand this a code a little bit. The 0.06 "magic" number is close to CP_AM / FFT_AM if that sparks any ideas. Also, my understanding is that separating the analog audio from the subcarriers comes for free when you subtract the conjugate of the lower sideband from the upper sideband. How does this relate to synchronization is your previous comment?

argilo, 2020-08-29:
> The 0.06 "magic" number is close to CP_AM / FFT_AM if that sparks any ideas.

I had the same thought, but performance was worse when I used CP_AM / FFT_AM there.

> Also, my understanding is that separating the analog audio from the subcarriers comes for free when you subtract the conjugate of the lower sideband from the upper sideband.

That's being done here:

https://github.com/theori-io/nrsc5/blob/bb41b4f922714d5fceb853f4611fb095086068ae/src/sync.c#L493-L503

I don't know whether this is the best place to do it; maybe it could be done pre-FFT.

My understanding is that the secondary and tertiary subcarriers are mirrored so that the sum of their upper and lower sidebands is a pure imaginary number, i.e. confined to the Q channel, and the analog audio occupies the I channel. It seems to me that the receiver must be correctly synchronized in phase to prevent the I channel from bleeding into the Q channel.

argilo, 2020-08-29:
The hacky way that I'm trying to stay phase-synchronized at the moment is to compute FFTs (one per symbol) and measure the phase of bin 128 (which should be the AM carrier). I then do a linear regression and adjust `st->phase` and `phase_increment` so that the AM carrier is held close to a phase of zero degrees. The fudge factor of 0.06 would suggest that something is off by a few degrees though.

It could be that a pre-FFT approach to tracking the AM carrier would work better.

argilo, 2020-09-05:
Attachment B of [the latest patent disclosure document for NRSC-5](https://www.nrscstandards.org/standards-and-guidelines/documents/standards/nrsc-5-d/ibiquity-letter-ip-disclosure-updated-110805.pdf) lists a number of patents describing details of AM reception, and most of these patents are already expired. They could be useful for building a better receiver. I'm having a read over https://patents.google.com/patent/US5878089 now.

argilo, 2020-09-05:
That patent describes an elegant way of performing pre-FFT phase synchronization: the design of the digital subcarriers is such that their Q component sums to zero over one data frame. If the Q channel does not sum to zero over one frame, it's because the analog signal is bleeding in due to phase error, and the sum indicates how much phase error is present.

I'll take a crack at implementing this technique soon.

argilo, 2020-09-08:
It turns out my post-FFT phase synchronization is already doing the same thing; it uses the zero bin of the FFT, and that's equal to the sum of the samples.

My next hypothesis is that the 0.06 fudge factor is just compensating for the lack of equalization in the receiver. The lower primary sideband is about 2 dB lower than the upper primary sideband in the KRLD recording, so the channel is clearly not ideal. Figure 4b in https://patents.google.com/patent/US6480536 shows what can happen in that situation: the AM sidebands bleed into the Q channel, interfering with the secondary & tertiary subcarriers.

I'll do a little more reading on equalization and see whether I can add it to the receiver.