theori-io/nrsc5 #243: Various unexpected block types with AM mode

pclov3r, 2020-08-19:
So doing some more testing I'm getting these unexpected block types. The sample is here https://gofile.io/d/FAYDGJ 

It's not perfect and drops sync but enough to do this.

```
23:50:26 Block @ 2
23:50:26 Sync!
23:50:26 ignoring partial pdu
23:50:26 crc mismatch!
23:50:26 crc mismatch!
23:50:26 BER: 0.067611, avg: 0.067611, min: 0.067611, max: 0.067611
23:50:26 Audio bit rate: 18.5 kbps
23:50:26 crc mismatch!
Unexpected block type: 3, expected 6
23:50:26 crc mismatch!
23:50:26 crc mismatch!
23:50:26 crc mismatch!
23:50:26 crc mismatch!
23:50:26 Lost synchronization
Unexpected block type: 1, expected 6
23:50:26 crc mismatch!
23:50:26 crc mismatch!
23:50:26 crc mismatch!
Unexpected block type: 4, expected 6
23:50:26 crc mismatch!
23:50:26 crc mismatch!
23:50:26 BER: 0.083454, avg: 0.075532, min: 0.067611, max: 0.083454
23:50:26 Sync!
23:50:26 ignoring partial pdu
23:50:26 Audio bit rate: 18.3 kbps
23:50:26 crc mismatch!
Unexpected block type: 4, expected 6
23:50:26 BER: 0.058074, avg: 0.069713, min: 0.058074, max: 0.083454
23:50:26 Audio bit rate: 18.5 kbps
23:50:26 BER: 0.057861, avg: 0.066750, min: 0.057861, max: 0.083454
23:50:27 crc mismatch!
Unexpected block type: 7, expected 6
23:50:27 crc mismatch!
23:50:27 crc mismatch!
Unexpected block type: 1, expected 6
23:50:27 Audio bit rate: 18.8 kbps
23:50:27 crc mismatch!
23:50:27 BER: 0.067602, avg: 0.066920, min: 0.057861, max: 0.083454
23:50:27 crc mismatch!
23:50:27 crc mismatch!
23:50:27 Audio bit rate: 18.8 kbps
23:50:27 BER: 0.061954, avg: 0.066093, min: 0.057861, max: 0.083454
23:50:28 Audio bit rate: 18.5 kbps
23:50:29 BER: 0.062241, avg: 0.065542, min: 0.057861, max: 0.083454
23:50:29 crc mismatch!
23:50:29 crc mismatch!
23:50:29 crc mismatch!
23:50:29 crc mismatch!
23:50:29 RS corrected 3 symbols
23:50:29 crc mismatch!
23:50:30 crc mismatch!
23:50:30 crc mismatch!
23:50:30 crc mismatch!
23:50:30 crc mismatch!
23:50:30 Lost synchronization
23:50:30 BER: 0.079139, avg: 0.067242, min: 0.057861, max: 0.083454
23:50:30 Sync!
```

pclov3r, 2020-08-24:
https://gofile.io/d/Zkgp7w another sample that's better then the previous.

awesie, 2020-08-24:
Does this second sample still have the crc mismatch message? I'm wondering if the "Unexpected block type" is due to packet corruption or not.

pclov3r, 2020-08-24:
Not bad conclusion. It's certainly possible.

A better sample would help a lot to rule that out.