theori-io/nrsc5 #251: Mac install

KF7ZVL, 2020-10-19:
I am trying to instal on a mac running 10.13.6. I am installing via Brew and get the following.

"Calling Non-checksummed download of nrsc5 formula file from an arbitrary URL is disabled! Use 'brew extract' or 'brew create' and 'brew tap-new' to create a formula file in a tap on GitHub instead."

Anyone know what I am doing wrong and how to fix it?

awesie, 2020-10-19:
This is caused by (recent?) changes to homebrew.

With some googling, using wget to first download the formula might be an easy "fix" (see https://dev.to/gjrdiesel/getting-around-brew-s-error-calling-non-checksummed-download-of-17fl).

argilo, 2020-11-26:
I updated the readme with instructions that should work with the latest homebrew: f964fb86c20daa867cacf02ac57ae9d3fc762abe

closed 2020-11-26 by argilo