theori-io/nrsc5 #253: [Windows] Static with -o to stdout

whothisbeme, 2020-11-27:
Hey there,

I'm not certain but I believe this is a bug encountered on the Windows Builds.

The below command on Ubuntu 18.04 will redirect **stdout** to the _testredirect.wav_ file.

``nrsc5 -o - 101.5 0 > _testredirect.wav``

This works on Ubuntu 18.04 and will provide a .wav file which opens in VLC and plays without issue.

If you run the same command using a Windows build it will redirect to the _testredirect.wav_ file but the produced file is un-usable.

When attempting to open via _Groove Music_ you encounter an error:
[image]

At first it seems you are unable to play this produced file but it seemed to be a header issue. I produced another .wav file using the following command:

``nrsc5 -o testsavetofile.wav 101.5 0``

This produced a file **_testsavetofile.wav_** that does play and works perfectly. I then compared the two files using HxD and it seems the broken file **_testredirect.wav_** has a "funky" header which I thought was causing the issues. I created a new file called **_franken.wav_** which uses the header from **_testsavetofile.wav_** and the data from **_testredirect.wav_**. This leaves me with a working .wav file which will play but has an extreme amount of static/noise but you can hear the station in the background.

I originally encountered this bug yesterday when attempting to run the following NodeJS code using the Windows build:

```
var wav = require('wav');
var Speaker = require('speaker');
var child_process = require('child_process');

var totalData = 0;

var outputWavWriter = new wav.FileWriter('./test.wav', {
    sampleRate: 44100,
    channels: 2
})

var wavreader = new wav.Reader();

const speaker = new Speaker({
    channels: 2,
    bitDepth: 16,
    sampleRate: 44100
})

// the "format" event gets emitted at the end of the WAVE header
wavreader.on('format', function (format) {
    console.log('Format event fired.');
    console.log(format);
    // the WAVE header is stripped from the output of the reader
    wavreader.pipe(speaker);
})

// START SCRIPT
run_script('nrsc5', ['-o', '-', '104.7', '0'], function(output, exitcode) {
    console.log('Process finished.');
    console.log(`Closing code: ${exitcode}`);
    console.log(`Output: ${output}`)
})

console.log(`Node Version: ${process.version}`);

console.log(`Starting NRSC5 NodeJS child_process testing...`);

function run_script(command, args, callback) {
    console.log('Starting Process.');
    var child = child_process.spawn(command, args);

    child.stdout.pipe(outputWavWriter) // Write to wav file
    child.stdout.pipe(wavreader); // Play right to speaker

    child.stdout.on('data', async function (data) {
        console.log(`Got Data: ${data.length}`);
        totalData += data.length;
        console.log(`Total Data: ${totalData}`)
    })

    child.stderr.on('data', function (data) {
        console.log(`stderr: ${data}`);
    })
    
    child.on('close', function(code) {
        callback('datatopassback', code)
    })
}
```

To run the above code save as index.js and run the following commands within an empty folder using NodeJS v14.15.1:

```
npm init -y
npm install wav speaker
node index.js
```

I ran the above NodeJS code using v14.15.1 on Windows first. 

It will play the audio over speakers and save a .wav file with the proper headers but the produced audio contains a lot of noise.

This morning I ran the same code but using a PC running Ubuntu 18.04 with the Linux build. The above NodeJS code and also the commands work perfectly under linux. The audio contains no noise and is clear.

I've tested this across _two_ Windows 10 PCs but using the same RTL2832U + R820T SDR receiver.

Is it possible I'm doing something wrong within my setup or maybe this is a known/expected output on Windows?

I've attached a _samples.zip_ which includes the produced files above for troubleshooting.
[samples.zip](https://github.com/theori-io/nrsc5/files/5609518/samples.zip)

Also, I've tried the MSYS2 64bit build, Ubuntu Cross-Complied mingw-w64 (64/32bit) builds both contain noise in the audio when using `-o -` in the command args.

Here are some screenshots of the Hex for a working .wav and the broken noisy .wav:

[image]

[image]

Thank you so much for this sweet open source software and all of the time invested,
Ryan Merritt

argilo, 2020-11-28:
I suspect this might be the cause:

https://github.com/theori-io/nrsc5/blob/b004fcaaca1a3475f3f291c038d5ec2f5b31e530/src/main.c#L699-L701

That is needed so that track titles containing non-ASCII characters display correctly.

Perhaps you could try removing that line to see if the problem goes away.

whothisbeme, 2020-12-02:
I've tested with the line removed and the noise is still present on 32/64bit Windows builds.

Are you able to duplicate the noise issue I'm encountering?

argilo, 2020-12-02:
Does the output file still look the same in your hex editor after that change? It could also be that newlines are problematic.

I don't often use Windows, so I haven't tested this. Any information you can gather would be helpful.

whothisbeme, 2020-12-02:
It does still look the same after that change. I removed the line, removed build-win32 folder and then ran ```support/win-cross-compile 32``` which I believe would be a clean build.

I've attached some screenshots of what I'm seeing via the hex editor and it looks fairly unchanged. Also here are the sample 
wav files.

[more_samples.zip](https://github.com/theori-io/nrsc5/files/5627653/more_samples.zip)

Original Redirect (Doesn't open, and if header fixed contains noise)
[image]
After Line Removal Redirect (Same, Doesn't open, and if header fixed contains noise)
[image]

Test Wav Original (No noise)
[image]
Test Wav After Line Removal (No noise)
[image]

I attempted to look through the source a bit more but so far I've not come across anything obvious however I've never written anything in C.

 I was thinking it could be something with the `cli.py` file and https://github.com/theori-io/nrsc5/blob/bb41b4f922714d5fceb853f4611fb095086068ae/support/cli.py#L91

However, after further thought I don't think its relevant considering it works perfectly fine on Linux so it must be something else with the way Windows is handling the stdout. Is it possible that it has something to do with binary/text modes when the file is opened on Windows? I don't believe Unix has this distinction between binary and text modes but I could be mistaken.

Thanks again for your help with this 👍

argilo, 2020-12-02:
Text mode does sound like a likely candidate.

Could you try adding `setmode(fileno(stdout),O_BINARY);` right after `SetConsoleOutputCP(CP_UTF8);`? If that works, we could probably do that conditionally if `-o -` is specified.

whothisbeme, 2020-12-02:
I attempted to use ```setmode(fileno(stdout),O_BINARY);``` right after ```SetConsoleOutputCP(CP_UTF8);``` but I was greeted by ```did you mean ‘REG_BINARY‘``` error.

I added `#include <fcntl.h>` & `#include <io.h>` after `#ifdef __MINGW32__` which resolved the error but after testing the produced redirect file is looking the same. It looks as if  ```setmode(fileno(stdout),O_BINARY);``` changed nothing with the output, but I'm not certain its working?

I'm starting to think we are on the right track. Is it possible we need to call ```setmode``` in another location? 

Or Is it possibly an issue with:

https://github.com/theori-io/nrsc5/blob/b004fcaaca1a3475f3f291c038d5ec2f5b31e530/src/main.c#L94

https://xiph.org/ao/doc/ao_open_file.html
https://xiph.org/ao/doc/drivers.html

Maybe the way Windows is treating stdout via libao? Do we need to call ```set_mode``` right before we send to stdout? I see it was recently [resolved ](https://gitlab.xiph.org/xiph/libao/-/merge_requests/1/diffs)to use Binary mode for file output on Windows but I don't see any mention of stdout other than that it may not be supported to write directly to it, IE: may result in broken files?

From AO Driver Docs: 

```Writing WAVs to stdout will result in broken files. Use either the raw or the au driver instead.```

However it seems the 'wav' driver is used on Linux as well and writing to stdout then redirecting to a file works without issue.

pclov3r, 2020-12-02:
On Windows what I've always done to write to wave file on Windows is `nrsc5 102.5 0 -o test.wav`

Does this solve the issue for you?

whothisbeme, 2020-12-02:
@pclov3r I understand that `nrsc5 102.5 0 -o test.wav` works perfectly fine on Windows. This issue is specifically related to _`stdout`_ on Windows and redirecting output to a file.

Thanks for your two cents though.

whothisbeme, 2020-12-02:
It seems possibly that the output is being polluted by quite a lot of `C2 A0` which I believe is the hex for the UTF8 encoding of a Non-breaking space.

I'm only showing 6 occurances of `C2 A0` in the hex from a working _`test.wav`_ file but the _`testredirect.wav`_ file contains 19,680 hits.

[image]
[image]

whothisbeme, 2020-12-03:
@pclov3r Can you possibly try ```nrsc5 -o - 101.5 0 > testredirect.wav``` with your build to see if the resulting **testredirect.wav** file is playable or not?

Thanks for your help and time spent investigating 👍

pclov3r, 2020-12-03:
That seems to have made no difference here.

Hum...

whothisbeme, 2020-12-03:
That is strange, what version of Windows are you running? 

So you are saying that the resulting **testredirect.wav** file is playable? Would you be able to provide a sample of the output with ```nrsc5 -o testsavetofile.wav 101.5 0``` and ```nrsc5 -o - 101.5 0 > testredirect.wav``` to compare?

Thanks again!

pclov3r, 2020-12-03:
@whothisbeme 

Sorry! I meant to say that it made no difference form the issue your having. Trying to play the file in VLC results in the same static audio that your having. :)

whothisbeme, 2020-12-03:
Ah gotcha, thank you for clarifying 👍

argilo, 2025-04-09:
I tested the `setmode(fileno(stdout),O_BINARY)` change myself, and it did fix the problem. So I think an appropriate fix would be to do that in the various places that write data to stdout (I/Q, audio, HDC) or read data from stdin (I/Q).

argilo, 2025-04-09:
#393 should fix this.

PR #393 (2025-04-09): Use binary mode for stdin and stdout on Windows
Fixes #253.

nrsc5 can write binary data to stdout, or read binary data from stdin:

* `-o -` writes audio to stdout
* `-w -` writes I/Q to stdout
* `--dump-hdc -` writes HDC to stdout
* `-r -` reads I/Q from stdin

But these don't work properly on Windows, since stdout and stdin use "text" mode by default, which automatically transforms line endings. Switching them to binary mode corrects the problem.

I've done so after the call to `parse_args`, since the `-v` option prints version information to stdout, and this should be done in text mode.
closed 2025-04-09 by argilo