theori-io/nrsc5 #254: win-cross-compile script missing LIBRARY_DEBUG_LEVEL variable

pclov3r, 2020-12-02:
It appears the win-cross-compile script missing is missing the LIBRARY_DEBUG_LEVEL variable

Would be useful if the script was edited to define the debug level without having to edit the script.

argilo, 2025-03-07:
How would you propose to make this work? The script currently passes all arguments after `(32|64)` through to `make`.

argilo, 2025-03-07:
Also, what debug messages do you find yourself needing? In theory, log messages from the library should only be needed for nrsc5 development, although some of them do currently display data which is not (yet) accessible through the API.

pclov3r, 2025-03-07:
What is mostly helpful is seeing the gain values set.  As you said they are helpful for debugging and development.

To make this work, Perhaps a if else statement to set a variable for the debug level that passes to make?

pclov3r, 2025-03-07:
I guess you could also use `read` to set a variable.

Tho, This would prompt the user to enter a debug level.

argilo, 2025-03-07:
If the gain information is useful to you as a user, I think it would be best to make it visible to API clients, and have the clients display it.

pclov3r, 2025-03-07:
> If the gain information is useful to you as a user, I think it would be best to make it visible to API clients, and have the clients display it.

That sounds like a good idea to expose some of that information via the API.

Perhaps a separate issue should be opened for that.

argilo, 2025-03-07:
Yes, I think it would make sense to open a separate issue (or perhaps multiple issues) for any information that would be useful to users or API clients, and that is not yet exposed through the API.

pclov3r, 2025-04-07:
Closed in light of https://github.com/theori-io/nrsc5/pull/389

closed 2025-04-07 by pclov3r