theori-io/nrsc5 #255: Is FC0012 supported? I get no valid PLL combination found

Prerunning, 2021-01-08:
Desktop@Desktop MINGW64 ~
$ nrsc5.exe 102.3 1
Found Fitipower FC0012 tuner
Exact sample rate is: 1488375.071248 Hz
Disabled direct sampling mode
[FC0012] no valid PLL combination found for 4293702179 Hz!
15:00:44 Set direct sampling failed.

andrewfer000, 2021-03-03:
This happens on my R820T2 as well. It's nothing serious.

argilo, 2025-03-01:
It appears the error message is printed when the direct sampling mode is switched off. Switching off this setting used to cause problems for E4000 tuners (#267), and so the nrsc5 command-line program was changed to set it only when the `-D` command line option is specified (#271). Thus I expect this error message should no longer appear.

closed 2025-03-01 by argilo