theori-io/nrsc5 #257: [Windows] error building with MSYS2

jnelson3, 2021-03-09:
Attempted to build with MSYS2. Followed all the instructions and got 
```
$ nrsc5/support/msys2-build
:: Starting core system upgrade...
 there is nothing to do
:: Starting full system upgrade...
 there is nothing to do
pacman: unknown option -- a
```
when running the build script.

FairyballYT, 2021-03-21:
What version of MSYS2 are you using to build nrsc5? Are you using the 64-bit or the 32-bit version of MSYS2? I had luck building nrsc5 with the 32-bit version of MSYS2.

searock35, 2021-04-28:
I had accidentally opened just MSYS instead of MSYS2 MinGW 32-bit on my computer. Opening the latter instead of the former solved this issue for me.

FairyballYT, 2021-04-29:
Oh, well then good luck!

FatRomanticSlob, 2021-05-17:
Had the same issue and then switched to mingw32.exe. However, nrsc5.exe doesn't appear to work. Double-clicking on the .exe file just flashes the screen momentarily and nothing happens.

jdjung21, 2021-12-27:
Hm, I'm having the same issue. I've tried compiling on both the 64 and 32 bit version and getting the same error. 

`juind@Eluktronics-The-Little-Monster MINGW32 ~
$ nrsc5/support/msys2-build
:: Starting core system upgrade...
 there is nothing to do
:: Starting full system upgrade...
 there is nothing to do
warning: autoconf-2.71-2 is up to date -- skipping
warning: automake-wrapper-11-2 is up to date -- skipping
warning: git-2.34.1-1 is up to date -- skipping
warning: gzip-1.11-1 is up to date -- skipping
warning: make-4.3-1 is up to date -- skipping
warning: mingw-w64-i686-gcc-11.2.0-5 is up to date -- skipping
warning: mingw-w64-i686-cmake-3.22.1-1 is up to date -- skipping
warning: mingw-w64-i686-libltdl-2.4.6-21 is up to date -- skipping
warning: patch-2.7.6-1 is up to date -- skipping
warning: tar-1.34-1 is up to date -- skipping
warning: xz-5.2.5-1 is up to date -- skipping
 there is nothing to do
Already up to date.
CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Building for x86_64-w64-mingw32
CMake Error at CMakeLists.txt:39 (message):
  Missing libtoolize.  Install libtool package and try again.

-- Configuring incomplete, errors occurred!
See also "C:/msys64/home/juind/nrsc5/build/CMakeFiles/CMakeOutput.log".

juind@Eluktronics-The-Little-Monster MINGW32 ~
`

Bweebo, 2022-01-09:
As it says, you need to install the libtools package. 

You can do that by entering 
$ pacman -S libtool

Let it install and then run the compile script again

argilo, 2023-07-04:
It looks like this issue was due to user error rather than a bug in nrsc5, so I'll close off this issue.

closed 2023-07-04 by argilo