theori-io/nrsc5 #260: Weird audio distortion on not-main channels

MB6, 2021-03-21:
Hello,
I've compiled nrsc5 on windows, on multiple pc's now, and when I try to tune not-main ie (1-3) subchannels, I get a weird distortion, grainy quality.

Here is a sample. https://mega.nz/file/jp1CQYjB#mCDIBDRvp3UqZIyKop3Zt2fRGqIIBRHRpxhpBv6inTc

64 bit build. I know my way around computers, but programming is not my forte. I'll provide anything else someone might need.

pclov3r, 2021-09-16:
I just noticed this now on a iHeart station subchannel here.

Is this a regression of https://github.com/theori-io/nrsc5/issues/212 however it appears that this bug was fixed?

@argilo Any opinion?

argilo, 2021-09-17:
It could also be a bug in faad2 or nrsc5's HDC patches. Unfortunately I know very little about the codec; @awesie might have more insight.

Roman-Port, 2022-01-25:
I'm also able to reproduce this. It sounds like it's dropping samples to me. Doesn't seem to affect HD-1, only HD-2 and beyond. I did some tests on my machine with the same IQ file. I built on both Windows and Ubuntu 20.02 to see if it was Windows related, but the output audio file matches byte-for-byte on the two platforms. 

Here's an example IQ recording. To reproduce, download this file and unzip it. Begin listening to HD-2 by invoking with command ``nrsc5 -r example_trim 1`` (NRSC5 starts subchannels at 0, their usual labels start at 1). After the ad and legal ID, you'll hear a song that has an especially garbled intro. Probably ~10 seconds into the song, press the 0 key to switch to HD-1 and compare it. The same ad will be playing on HD-1 and even that sounds clearer than the HD-2 channel. Once a song begins, it's a night and day difference.

https://drive.google.com/file/d/1DBqvxH8wEjrWa3nGDIFPF5TGWAvQRLhb/view?usp=sharing

Below, I've also uploaded the raw HE-AAC audio data for both HD-1 and HD-2. Note that these won't be playable without the patch in this repo.

[KPFX-HD1.aac](https://github.com/theori-io/nrsc5/files/7931052/KPFX-HD1.zip)
[KPFX-HD2.aac](https://github.com/theori-io/nrsc5/files/7931053/KPFX-HD2.zip)

TheDaChicken, 2022-02-24:
I am also able to reproduce this, especially on iHeart subchannels! A station here got dropped to a subchannel. The only way for me to listen to it is on their site or HDRadio 2 subchannel! It sounds terrible listening to it on the hdradio 2 subchannel using this program! 
I know nothing about HDC codec. I would've tried to find the issue if I knew about the codec.

Roman-Port, 2022-02-24:
Any chance someone with a "real" HD Radio with a digital output could record a station's HD-1 and HD-2 alongside an IQ recording for use with nrsc5? That way we could compare, byte-for-byte, the output of the patched codec to the known-good output of a real radio. Since it's all digital, the outputs from both should match exactly. Unfortunately, my Sony XDR-F1HD only has an analog output so I can't do it myself. Tomorrow I'll get in touch with an engineer friend of mine to see if I could borrow one of his sets with a digital output.

ferrellsl, 2022-02-24:
I suspect this issue is related to your dongle(s) and their ppm rate(s).  I have 3 dongles, two cheap ones by Nooelec and one expensive stick from RTL-SDR.COM.  After measuring the PPM of these 3 sticks using SDR#, the dongle from RTL-SDR.COM had a ppm rate of -1 which is fine but the cheaper dongles from Nooelec had a ppm of -7 and -12 after running for about half an hour.  I recommend using SDR# to measure the ppm of your sticks after they get good and warmed up and then use the -p <ppm error> switch with the nrsc5 exe to see if your problem persists. It should go away as it did with my cheap sticks.
-- The quality of your thoughts will determine the quality of your life.

Roman-Port, 2022-02-24:
I don't believe that this would be related to the dongle. If this were an analog signal I could believe it, but with a digital signal like this we would be seeing very different problems. The program never loses sync, and all frames check out with their checksum, so we know everything was received correctly. This seems like it's simply a problem with decoding the valid digital information.

ferrellsl, 2022-02-24:
There is no such thing as a digital signal in the world of radio.  All radio signals are analog, including the ones picked up by your dongle as well a as iHeart HD Radio transmissions. Digital info is modulated onto the analog carrier wave.  If you're not tuned correctly to the proper freq. because of receiver drift found in these cheap dongles, your ppm error rate will be high. This will cause you to experience the problems you're describing.  You need to measure the ppm error rate of your dongles and adjust accordingly.  I have 3 of these dongles running on 3 different PCs running a mix of Linux and Windows and they all run properly after adjusting for the ppm error rate.
It's called HD (Hybrid Digital) Radio for a reason.  Hybrid, because an analog carrier is used to modulate digital information.
See:  www.britannica.com/technology/digital-signal-modulation

| 
| 
| 
|  |  |

 |

 |
| 
|  | 
digital signal modulation

Other articles where digital signal modulation is discussed: telecommunication: Digital modulation: In order ...
 |

 |

 |

-- The quality of your thoughts will determine the quality of your life.

Roman-Port, 2022-02-24:
Thanks for the response. I understand that it's underlying modulation is analog, but that signal is demodulated into a digital stream. However, we know that step was performed correctly because all of the checksums used to detect errors pass. This guarantees that all of the digital data was successfully received.

awesie, 2022-02-24:
Let's avoid unhelpful comments.

Thanks @Roman-Port for the example IQ recording. I think it is likely a problem with the HDC decoding for your example. If you listen to the left and right channels separately, it sounded to me like only one of those channels is distorted. Do you agree?

ferrellsl, 2022-02-24:
You're not understanding the problem.  The ppm error rate/receiver drift will cause exactly the problems you're describing despite your analysis of digital checksums.  The problem you're describing is occurring BEFORE any demodulation occurs so it doesn't matter that the checksums you're looking at appear to be correct.  So no, the checksums don't guarantee that all of the digital info was successfully received because you're not tuned to the center of the carrier frequency.  You are capturing only portions of the demodulated data due to poor tuning (most likely cause by receiver drift), and the checksums for those portions will appear to be correct or they wouldn't even be decoded.
You can keep insisting that the software is at fault, but until you test your dongle(s) for receiver drift and figure out how far to the left or right of the carrier it/they drifts after heating up, then no one here is motivated to help you since only 2 people here have this problem.  It can also be caused by a weak signal or intermittent interference.  You need to fire up SDR# or other RTL-SDR software and tune to a carrier and then wait about half an hour to see how far the receiver drifts from the carrier.  Cheap dongles are notorious for this problem.

And before you start questioning my experience in the world of radio, I was a career Army Signal Officer who retired as a Lieutenant Colonel after 23 years of service and multiple combat deployments.  I understand how NRSC5 works quite well in addition to the hybrid digital radio systems in use by the US military.
-- The quality of your thoughts will determine the quality of your life.

TheDaChicken, 2022-02-24:
> I suspect this issue is related to your dongle(s) and their ppm rate(s).  I have 3 dongles, two cheap ones by Nooelec and one expensive stick from RTL-SDR.COM.  After measuring the PPM of these 3 [...]

I am using an expensive stick from RTL-SDR.COM.

TheDaChicken, 2022-02-24:
> Thanks @Roman-Port for the example IQ recording. I think it is likely a problem with the HDC decoding for your example. If you listen to the left and right channels separately, it sounded to me like only one of those channels is distorted. Do you agree?

I would agree one of the channels is distorted. When I listen to a subchannel (2-3), it does sound like only one of those channels is distorted.

ferrellsl, 2022-02-24:
Have you measured your dongle's PPM rate?  If so, what is it?  Also, you need to ensure that your MER is ideally above 5.0dB. Higher is better.  If it drops below that value or bounces up and down significantly you'll experience breaks in the data stream which will cause breaks in the audio.
One other issue that I've experienced that causes choppy audio is confined to just one of my systems running Windows.  For some reason when using the WinUSB driver installed by Zadig, I got intermittent signal loss.  I switched to the libusb0 driver and that fixed it.

-- The quality of your thoughts will determine the quality of your life.

Roman-Port, 2022-02-24:
> > Thanks @Roman-Port for the example IQ recording. I think it is likely a problem with the HDC decoding for your example. If you listen to the left and right channels separately, it sounded to me like only one of those channels is distorted. Do you agree?
> 
> I would agree one of the channels is distorted. When I listen to a subchannel (2-3), it does sound like only one of those channels is distorted. Though I do want to say that the frequencies I am getting this issue on, the first HD1 channel is 40kbps. I do get a higher bitrate on other stations? Then when I switch to HD2-3, the bitrate goes to 30kbps.

Yeah, typically the HD2+ stations (referred to as SPS channels) are at a lower data rate than the HD1 channel. However, the same kind of distortion isn't present if I listen using a "real" HD radio like my Sony XDR-F1HD so we know it's a problem on our end.

ferrellsl, 2022-02-24:
Yes, that's certainly a possibility too.  Here in Phoenix we have several stations with content on the secondary and tertiary channels where the broadcasters are using a low bit rate of around 20kbps.  That's fine for talk radio but any music at that low of a bit rate sounds distorted.
Do the second and 3rd channels on stations using higher bit rates, say 30kbps or higher sound distorted too?
-- The quality of your thoughts will determine the quality of your life.

argilo, 2022-02-24:
PPM error in the receiver's oscillator is not a problem for nrsc5, because it compensates for error (within the range of approximately -125 to +125 PPM) and precisely tracks any frequency drift.

If BER values are at or near zero and no checksum errors appear, then the problem almost certainly lies in the HDC decoder.

ferrellsl, 2022-02-24:
Thanks for educating me.  I didn't realize that the binary auto compensates for ppm.  Should probably remove the -p <ppm error> switch then unless you foresee some users with receivers that fall outside the -125 to +125 rate.
I think he's just listening to some secondary and tertiary stations using very low bit rates.
-- The quality of your thoughts will determine the quality of your life.

MB6, 2022-02-24:
Hello! OP here

I'm an audio engineer, I know what AAC compression sounds like. That sound in my initial sample is not it. That does not rule out other people hearing the compression, but I do not that that was my initial problem.

logiconcepts819, 2022-09-10:
Here were some observations I noticed in the libfaad2 implementation:

- The HDC frame length is set at 1,024 samples, which means synthesis works on 2,048 audio samples at a time.
- The HDC codec in libfaad2 utilized the DRM PS decoder, which works on 30 subsamples of QMF data with 64 QMF channels per subsample, for a total of 30*64 = 1,920 samples.
- Synthesis produces 2,048 samples, but PS decoding seems to work only with 1,920 samples.  This difference seems to be the main culprit of the distortion in the stream when the DRM PS decoder is used.

After decreasing the frame length to 960 so that synthesis works on 1,920 samples at a time, the overall audio quality was much better.  However, there were a lot of underruns when streaming live, and the audio sounded sped up as well, probably by (2048/1920 - 1)*100% = 6.67%.

This seems to indicate two things.  First, the frame length should be kept at 1,024.  Second, a new PS decoder implementation for the HDC codec could be necessary, possibly with QMF constants computed for 2048/64 = 32 subsamples instead of 30.  Just my two cents.