theori-io/nrsc5 #262: Frequent loss of synchronization on SDRPlay?

andrei-ch, 2021-05-12:
I am running nrsc5 on macOS using SDRPlay (RSP1A). Overall it's been pretty successful and I'm getting short FM HD programs correctly decoded using rsp_tcp for IQ server:

rsp_tcp -a 127.0.0.1 -f 89500000
nrsc5 -H 127.0.0.1 89500000

The problem is that after several minutes into session, nrsc5 starts to lose synchronization frequently, which eventually results in unlistenable stream. FM reception seems to be pretty stable in the area on all devices. Loss of sync happens with all stations, regardless of distance to transmitter.

I also built a tuner using RSP APIs and nrsc5 library. It has the same problem.

Any tips how to diagnose the issue? Does it happen with other hardware?

argilo, 2021-05-12:
There are a number of things that could cause this problem. The first thing to check would be whether your RF gain is set too high or too low. The MER and BER values reported by nrsc5 should give some indirect indication of whether this might be a problem. If it's not that, then I'd suggest making an I/Q recording using nrsc5's `-w` option and sharing that.

markjfine, 2021-05-12:
Unsure if this is the same in SDRPlay, but have notice with an RTL-SDR v3 that it tends to get real hot, real fast. Have suspected for some time that this creates de-tuning problems as well as noise level/intermod issues over time. Some of this can be offset by adjusting the RF gain, but it's possible that a ppm adjustment may also be necessary. Best way to do that is to use a regular FM tuning package with a visual display and let it run over time to see if it de-tunes, then apply /note whatever ppm adjustment may be needed to make it more preceise.

andrei-ch, 2021-05-13:
The issue was a combination of several things. Better antenna and removing Wi-Fi interference stabilized rsp_tcp + nrsc5 workflow.

The rest of the problem was caused by handling of AGC in SoapySDR. I suspect a bug there, but every time gain changed, input stream got disrupted enough for nrsc5 to lose synchronization. I stopped investigating there and opted to rewrite input stage using SDRPlay API directly which should take care of AGC and a few other things.

Thanks for the tips!

closed 2021-05-13 by andrei-ch