theori-io/nrsc5 #267: Allocating 8 zero-copy buffers

bbarcee, 2021-12-23:
When I run the command nrsc5 102.5 0 or any channels /programs that I can see have the digital side bands it would see the command hangs 

Found Elonics E4000 tuner
Exact sample rate is: 1488375.071248 Hz
Disabled direct sampling mode
Allocating 8 zero-copy buffers

and then I get a blinking block, no further output  but I can play the test file
I'm using linux mint and Ive tried using both of these -DUSE_SSE=ON and -DUSE_FAAD2=ON cmake options (and each separately) with the same output. Ideas?

bbarcee, 2021-12-24:
I figured it out! 

I tried uninstalling and reinstalling making sure I followed the instructions... And I could use the E4000 with commands from rtl_fm but nrsc5 would sit there and try to start the stream and fail. the E4000 tuner is inside the NooElec nesdr Smartee radio (the silver one). 

I deiced to try out my nesdr smart (black one) and it shows up as the Found Rafael Micro R820T tuner when accessed by nrsc5... it allocated, and WAM BAM, MER, BER love! 

Should I delete these posts and create one with the finding? I will try to get the other unit working too.

argilo, 2022-01-31:
Does it make any difference if you manually specify a gain value (e.g. `-g 20`)?

I don't have an E4000 tuner handy, so unfortunately I won't be able to test directly.

jboone, 2022-02-10:
I also have a NooElec NESDR SmartXTR (E4000-based) that does almost the same thing, except I don't get the "Allocating..." message. I get just:

```
$ nrsc5 -l 1 -g 49 -o - 92.3 0    # Just one example, as I got desperate to find options that changed behavior.
Found Elonics E4000 tuner
Exact sample rate is: 1488375.071248 Hz
Disabled direct sampling mode
```

And then it seems to hang. I've tried increasing logging (`-l 1`), explicitly setting gain (`-g 49`), sending output to stdout (`-o -`), and different frequencies and programs, all of which are claimed to be HD Radio. No difference in behavior.

I do believe I compiled with `SSE=ON`, if that's interesting.

I got out a NooElec NESDR SMARtee (R820T-based), and it more-or-less fired right up.

```
$ nrsc5 -l 1 91.5 0
Found Rafael Micro R820T tuner
Exact sample rate is: 1488375.071248 Hz
[R82XX] PLL not locked!
Disabled direct sampling mode
[R82XX] PLL not locked!
12:33:50 Synchronized
12:33:50 Country: US, FCC facility ID: 50607
12:33:50 Station name: KOPB-FM
```

On another station, it did sit at the second `[R82XX] PLL not locked!` for a while and I got impatient and quit (ctrl-C) before I tried the station in the example above.

argilo, 2022-02-12:
I was able to get my hands on an E4000-based RTR-SDR and reproduced the problem. Commenting out this code solves the problem:

https://github.com/theori-io/nrsc5/blob/d38f05a3ae1ecde1f6d9404c73a6ba618b6daa5b/src/nrsc5.c#L451-L453

It could be that there's a bug in librtlsdr's `rtlsdr_set_direct_sampling` function, where turning off direct sampling mode doesn't work on E4000-based devices.

argilo, 2022-02-12:
I guess a suitable workaround might be to call `rtlsdr_set_direct_sampling` only if the `-D` command line option is specified.

argilo, 2022-02-12:
I opened #271, which should fix this. If you have a chance to test, please let me know how it works @bbarcee @jboone.

jboone, 2022-02-12:
Retested:

```
$ nrsc5 -v
nrsc5 revision d38f05a
$ nrsc5 103.3 0
Found Elonics E4000 tuner
Exact sample rate is: 1488375.071248 Hz
Disabled direct sampling mode
# Hangs
```

Checked out the `fix-direct-sampling-e4000` branch and rebuilt.

```
$ nrsc5 -v
nrsc5 revision 1754b3e
$ nrsc5 103.3 0
Found Elonics E4000 tuner
Exact sample rate is: 1488375.071248 Hz
# Hangs
```

Swapped out the antenna, just in case... Tried specifying direct sampling mode (which, given what you said, probably wouldn't help anyway).

```
$ nrsc5 -D 0 92.3 0
Found Elonics E4000 tuner
Exact sample rate is: 1488375.071248 Hz
Disabled direct sampling mode
# Hangs
^C
$ nrsc5 -D 1 92.3 0
Found Elonics E4000 tuner
Exact sample rate is: 1488375.071248 Hz
Enabled direct sampling mode, input 1
# Hangs
^C
$ nrsc5 -D 2 92.3 0
Found Elonics E4000 tuner
Exact sample rate is: 1488375.071248 Hz
Enabled direct sampling mode, input 2
# Hangs
^C
$ nrsc5 92.3 0
Found Elonics E4000 tuner
Exact sample rate is: 1488375.071248 Hz
14:32:14 Synchronized
14:32:14 Country: US, FCC facility ID: 21
14:32:14 Station name: HD  
14:32:14 Country: US, FCC facility ID: 21
14:32:14 Station name: HD  
14:32:14 Audio program 0: public, type: Classic Rock, sound experience 0
14:32:14 Audio program 1: public, type: None, sound experience 0
# Well huh!
```

Disconnected the RTL-SDR from the laptop USB, waited ten seconds, reconnected.

```
$ nrsc5 103.3 0
Found Elonics E4000 tuner
Exact sample rate is: 1488375.071248 Hz
14:35:06 Synchronized
14:35:06 Data service: public, type: Navigation, MIME type 469
14:35:06 Data service: public, type: Image Maps, MIME type 2d7
14:35:06 Slogan: K103
14:35:06 Data service: public, type: Navigation, MIME type 469
14:35:06 Data service: public, type: Image Maps, MIME type 2d7
14:35:06 Country: US, FCC facility ID: 68210
# Nice.
```

Maybe I was just impatient waiting for the signal to sync up? Reattaching the dongle again, to double-check, waiting longer this time, just for fun.

```
$ nrsc5 103.3 0
Found Elonics E4000 tuner
Exact sample rate is: 1488375.071248 Hz
14:37:02 Synchronized
14:37:02 Data service: public, type: Navigation, MIME type eb2
14:37:02 Data service: public, type: Navigation, MIME type 469
14:37:02 Slogan: K103
14:37:02 Data service: public, type: Navigation, MIME type eb2
14:37:02 Data service: public, type: Navigation, MIME type 469
14:37:02 Country: US, FCC facility ID: 68210
# Waited for audio output this time. Yep, works!
```

Backing out the change and replugging the dongle, just to make sure it's not my impatience.

```
$ nrsc5 -v
nrsc5 revision d38f05a
$ nrsc5 103.3 0
Found Elonics E4000 tuner
Exact sample rate is: 1488375.071248 Hz
Disabled direct sampling mode
# Hangs. Waited an *entire* minute, which should've been plenty of time to sync with the broadcast signal.
```

And checking out and building the fix again...

```
$ nrsc5 -v
nrsc5 revision 1754b3e
$ nrsc5 103.3 0
Found Elonics E4000 tuner
Exact sample rate is: 1488375.071248 Hz
14:45:25 Synchronized
14:45:25 Data service: public, type: Image Maps, MIME type 2d7
14:45:25 Data service: public, type: Image Maps, MIME type e96
14:45:25 Slogan: K103
14:45:25 Data service: public, type: Image Maps, MIME type 2d7
14:45:25 Data service: public, type: Image Maps, MIME type e96
14:45:26 Country: US, FCC facility ID: 68210
# Works!
```

Verdict: User impatient after applying fix. Looks like your fix works!

argilo, 2022-02-12:
Glad to hear it! I've also done some testing of my own, and this appears to work well with both E4000 and R820T2 tuners. I'll merge the PR.

closed 2022-02-12 by argilo

bbarcee, 2022-02-14:
FYI I recompiled today and can confirm its working as expected - and really well, thanks! That smartee xtr gets HOT!

Appreciate both of you working on this one!

PR #271 (2022-02-12): Only call nrsc5_set_direct_sampling if -D is specified
Fixes #267.

It appears that calling `rtlsdr_set_direct_sampling(st->dev, 0)` on an E4000 tuner does not work properly. To avoid the problem, we can simply avoid calling this function at all if the `-D` command line option is not used, rather than defaulting to setting the direct sampling mode to zero.