theori-io/nrsc5 #268: Is there a way to dump compressed audio in a playable format? (without re-encoding)

salamanders, 2022-01-23:
The original post says "We found that it is essentially HE-AAC with some modifications".

I wasn't sure if it was possible to output HE-AAC files in a mp4 wrapper, without having to decode-to-wav-reencode-to-AAC.

icb-, 2022-01-24:
You can output the HDC stream with the `--dump-hdc <path>` option. Unfortunately, this isn't directly playable with unpatched decoders.

There used to be code to convert that stream to a standard ADTS, but it was [removed](https://github.com/theori-io/nrsc5/commit/0204afb9a1be323e86ce8c8c8efc46368465cba8) as "unsupported". It's still in the Git history, and you can build it. See a [comment on issue 88](https://github.com/theori-io/nrsc5/issues/88#issuecomment-336329564) for how to get it and build it. This seems to be mostly playable by software like VLC and ffmpeg, but may have some decoding and audio issues.

argilo, 2025-03-07:
I'll close this off, since the question was answered.

I suspect it won't be possible to revive the conversion feature, because I think the differences between HDC and HE-AAC go beyond just how the bits are packed.

closed 2025-03-07 by argilo