theori-io/nrsc5 #269: wont launch

borgom7615, 2022-01-26:
I'm sure this is a simple issue but i cant find a previous ticket that highlights this.

compilation worked, with the msys mingw32. the demo commands post completion within mingw32 worked as well. but when i go to launch NRSC5,EXE either within the msys folder or copied with the correct application extension else where, it dose not launch, it flash's up and away on the tool bar within a moment.  if I try enough times it opens but the moment I go to type anything it closes immediately. 

i did have to install libtools to compile, that's not something mentioned in the windows msys instructions. perhaps I was not to do that?

any help would be apricated thanks! 

attached is the exe and dll that I copied into a different folder as suggested by the guide.

[nrsc5.zip](https://github.com/theori-io/nrsc5/files/7943878/nrsc5.zip)

argilo, 2022-01-26:
Are you running nrsc5.exe from the command line (i.e. from within cmd.exe)?

argilo, 2022-02-12:
I'll close off this issue, since there doesn't appear to be any issue with nrsc5 itself.

closed 2022-02-12 by argilo