theori-io/nrsc5 #274: demuxing from hackrf

parity-error, 2022-03-01:
I  have a (20MHZ ) bit stream of raw IQ data coming in to my pipeline (its currently demodulating and recording all of the fm channels) and I want to take that same stream and individually shift it (like I do currently for the analog) then decimate and send it to nrsc5 for each digital channel that is present. 

I have been playing around with trying this using CSDR (https://github.com/ha7ilm/csdr) something like this:

(in this example the sample is taken with the center at the frequency that I want to test so I don't need to 'slide' (shift_addition_cc) it.
`cat 20Mhz.sample.fom.hackrf.iq | csdr convert_u8_f | csdr fir_decimate_cc 92 0.01 HAMMING | csdr convert_f_u8 | ./nrsc5/src/nrsc5 -r - 0`

But I don't get any output from nrsc5.   This seems theoretically possible  but I am not sure if the problem is a slightly different format of the IQ data or if I am not getting it to the exact sampling rate that nrsc5 is expecting.     Please advise.   Any help on this would be greatly appreciated.

icb-, 2022-03-02:
You want a [1488375 sample per second ](https://github.com/theori-io/nrsc5/blob/66ea75c8c4d2db4693570ebd2aaacecae1a774ce/src/defines.h#L12)interleaved uint8 stream.

This GNU Radio flow graph generates a playable IQ file from my Pluto SDR.
[image]

argilo, 2022-03-03:
Correct, the input sample rate needs to be exactly 1488375. A decimation factor of 92 will give you 217391 instead, which is way off. Also, I believe the HackRF produces signed 8-bit samples, whereas RTL-SDR tools (including nrsc5) use unsigned 8-bit.

argilo, 2022-03-03:
In case it's any more convenient for your use case, nrsc5's API (accessible from C or Python) has an `nrsc5_pipe_samples_cs16` function that accepts signed 16-bit complex samples at 1488375/2 samples per second (i.e. half the rate that the command line tool uses).

argilo, 2022-03-03:
Also, the Python command-line tool (cli.py) has an `--iq-input-format` option which allows you to choose between `cu8` and `cs16` formats. The sample rates are as above.

argilo, 2025-03-07:
It appears the issue here was outside nrsc5, so I'll close this off now.

closed 2025-03-07 by argilo