theori-io/nrsc5 #275: Logs to stderr, so can get WAV file output in standard output?

salamanders, 2022-03-20:
[yellowShoes](https://github.com/evuraan/yellowShoes) is somehow able to stream the output (decompressed WAV file) to standard output, yet is able to capture the log info which contains the song, artist, etc.

When I tried routing the `-o -` (route output to dash), it co-mingled the log lines with the binary output.  

Is there a way to route the log lines to somewhere else (like the error stream), allowing both the log and the WAV to be streamed out of the binary?

argilo, 2022-03-20:
The logger does in fact send its output to stderr:

https://github.com/theori-io/nrsc5/blob/1912b942ced009e8279326313e29fb9f812df6d0/src/log.c#L88-L100

What exactly are you seeing mixed into stdout?

salamanders, 2022-03-20:
Oh.  Oh no.  (looking at my code).  Yes you are totally correct and I had my code switched up and you just saved me so much debugging time.  THANK YOU!!

closed 2022-03-20 by salamanders