theori-io/nrsc5 #277: Successful Android Build

ferrellsl, 2022-05-18:
Install termux for Android.  I got it from the Google Play Store.

Termux has a package manager called pkg.  To install nrsc5 prerequisites, just run pkg for the missing packages.
i.e.: pkg install git cmake curl build-essential cmake autoconf libtool
Install VIM while you're at it because you will need to make two small changes to main.c

To access your SDR dongle, make sure you've installed termux-usb by invoking:

pkg install termux-api libusb clang
Let's assume the device is /dev/bus/usb/001/002. Ask for permission to access it:
termux-usb -r /dev/bus/usb/001/002

libao will have to compiled from source

download the sources for libao: curl -o libao-1.2.0.tar.gz https://ftp.osuosl.org/pub/xiph/releases/ao/libao-1.2.0.tar.gz
tar -vxzf libao-1.2.0.tar.gz
cd libao-1.2.0
./configure --prefix=$PREFIX
make
make install

For librtlsdr, run this script found here (Ham gear for Termux):  https://pastebin.com/xatGjWc9
This script will also install and build a few other libraries such as fftw

From here forward, follow the Linux build instructions found on the github project page.

After running cmake and make, the nrsc5 libraries will be built but main.c will fail to compile for two reasons.

1.  On line 811 there is a call for pthread_cancel.  I tried a couple hacks to work around that but no joy, I was in a hurry so I merely commented out the line with two // comment marks.

2.  The compiler will also fail to find the includes for libao.  As a quick and dirty hack I just copied the files ao.h and os_types.h from the libao include directory into the nrsc5/src folder and edited line 16 in main.c to read:  #include "ao.h"

Running make again should result in a successful build of the nrsc5 binary.

Running the binary without arguments results in the normal list of command line options.

On my un-rooted tablet, I get the following errors when I try to play back a station:  "Unable to open audio device" and "Open device failed." 

I'm assuming this is because I don't have root access on my Samsung Galaxy Tab S7+.  I also have to ctrl-c out of the termux-usb command when I ask for a list of devices.

Anyway, when I get the chance, I will attempt a build on a rooted device and see if that makes a difference.
Eventually it would be nice to have an Android app for HD Radio similar to the DAB-Z/DAB+ app in the play store for using a DAB dongle on Android.  See:  https://play.google.com/store/apps/details?id=com.zoulou.dab

drewinchas, 2023-07-25:
I followed these instructions and built nrsc5 successfully on my non-rooted tablet. USB device access does not work, but it can access an rtl_tcp server and it works properly. 

The Signalware "SDR Driver" app provides a local RTL TCP server that NRSC5 can use:
https://[play.google.com/store/apps/details?id=marto.rtl_tcp_andro](https://play.google.com/store/apps/details?id=marto.rtl_tcp_andro)

ferrellsl, 2023-07-26:
> I followed these instructions and built nrsc5 successfully on my non-rooted tablet. USB device access does not work, but it can access an rtl_tcp server and it works properly.
>
> The Signalware "SDR [...]

I finally got around to building this on a rooted tablet.  One of the issues that was preventing access to USB devices from within termux was that I was using the wrong version.  Don't use the termux found on the PlayStore.  Install f-droid and use it to install termux and termux-api from the f-droid repos.  Once that's completed, open termux and do a package update and then run: pkg install termux-api.

You should now be able to see your USB devices by typing:  termux-usb -l

After successfully building nrsc5 on my rooted tablet I can successfully capture audio streams to a file via my dongle.  For some reason when I try to play thru the speakers I get the following:  Unable to open audio device.

My device is rooted with Magisk.  To invoke nrsc5 and capture the stream to a file, go to the nrsc5/build/src folder and type
su -c ./nrsc5 94.5 0 -o test.wav

94.5 is the station and the 0 is for the 1st channel
 
[image]

I'm so close to getting this to stream to my speaker that I can taste it!  LOL!

Any help would be appreciated!

ferrellsl, 2023-07-26:
Made a little more progress.  Installing play-audio in termux and then installing mpv, I can play the captured audio files.  Maybe some bash shell gurus can figure out how to pipe the wave file from nrsc5 to mpv.  I'm out of my comfort zone when it come to writing bash scripts.

ferrellsl, 2023-07-26:
My first post is a bit outdated and the pastebin script is broken so I thought I'd write an updated post for building.

These instructions assume you have a rooted Android device.

I've probably missed a step or two along the way but this should be pretty complete.

Install f-droid and then install termux and termux-api from the f-droid repos. Don't use the termux found in the PlayStore.
Open termux and do a pkg update, then do a pkg install termux-api
You should now be able to see your USB devices by typing: termux-usb -l

Next, install the following packages: pkg install git cmake curl build-essential autoconf libtool binutils libusb tsu pulseaudio portaudio fftw

Install an editor to make changes to a couple of files later.  I prefer VIM.
pkg install vim

Download the sources for libao: 
curl -o libao-1.2.0.tar.gz https://ftp.osuosl.org/pub/xiph/releases/ao/libao-1.2.0.tar.gz

Build libao with the following:

tar -vxzf libao-1.2.0.tar.gz
cd libao-1.2.0
./configure --prefix=$PREFIX
make
make install

Don't use the pastebin script that I linked in my first post as it's now broken.

Next, download and install librtlsdr to your home directory:

git clone https://github.com/librtlsdr/librtlsdr.git
cd librtlsdr
cd src
vim rtl_adsb.c  (Delete or comment line 506: pthread_cancel(demod_thread); Save the file and exit vim)
cd ..
mkdir build
cd build
cmake -DCMAKE_INSTALL_PREFIX=$PREFIX ..
make
make install

Now we can clone the NRSC5 repo.  Go back to your home directory.

git clone https://github.com/theori-io/nrsc5.git
cd nrsc5/src
vim main.c  (Delete or comment line 832:  pthread_cancel(input_thread);  Save the file and exit vim)
cd ..
mkdir build
cd build
cmake -DCMAKE_INSTALL_PREFIX=$PREFIX -DUSE_NEON=ON ..
make 
make install

cd src
You should see an nrsc5 and a libnrsc5.so binary file in this folder.

I'm using Magisk for super user access so if you're using supersu or another app for root access, your mileage may vary.
To test your binary at this point, type:

su -c ./nrsc5 freq channel_numner -o test.wav

If all is working, you should end up with a wave file that can be played back using ffmpeg or mpv.
Press press 'q' to halt nrsc5 and then play your wav file.
.
I installed mpv and termux's play-audio packages to play the wav file.
pkg install mpv play-audio

Then, to play the captured audio, I just typed:  mpv test.wav

After some digging I've determined that the pulseaudio used by termux is a modified version that won't work with the nrsc5 binary, so this means we're stuck with streaming the nrsc5 output to a file.  Termux has its own versions of mpv and ffmpeg for playing audio which work well so it should be possible to pipe the file output from nrsc5 to them in real-time but I'm not a bash guru.

drewinchas, 2023-07-26:
> I finally got around to building this on a rooted tablet. One of the issues that was preventing access to USB devices from within termux was that I was using the wrong version. Don't use the termux [...]

Have you verified that your libao is building against and including pulseaudio support? I ask because the ao_pulse.c is missing '#include <time.h>.' Without that statement, the compiler will fail to find nanosleep() when it tries to build. If pulseaudio support isn't included, then you won't see this error.

drewinchas, 2023-07-27:
Here's my steps to build NRSC5 on Android **with working audio output:**

The key difference is to use the **libao package from Termux,** instead of rebuilding it:

1. Download Termux from Github (not the Play Store... I haven't tried F-Droid):
    [[https://github.com/termux/termux-app/releases/tag/v0.118.0](https://github.com/termux/termux-app/releases/tag/v0.118.0)](https://github.com/termux/termux-app/releases/tag/v0.118.0%5d(https:/github.com/termux/termux-app/releases/tag/v0.118.0)) 
2. Upgrade packages:
`pkg upgrade`
3. Install first set of dependencies (based on @ferrellsl's original post) **AND libao **:
`pkg install git cmake curl build-essential autoconf libtool binutils libao termux-api libusb fftw`
4. Clone the librtlsdr git repo (from the pastebin script):
`git clone https://github.com/librtlsdr/librtlsdr.git`
11. Change to librtlsdr source directory:
`cd librtlsdr`
12. Remove the pthread_cancel from rtl_adsb.c (we're not using ads-b with nrsc5 anyway):
`sed -i 's/pthread_cancel/\/\/pthread_cancel/g' src/rtl_adsb.c`
13. Build and install librtlsdr
`mkdir build && cd build`
`cmake -DCMAKE_INSTALL_PREFIX=$PREFIX ..`
`make`
`make install`
`cd ../..`
14. Clone the nrsc5 repo from git:
`git clone https://github.com/theori-io/nrsc5.git`
15. Change to nrsc5 directory:
`cd nrsc5` 
16. Comment-out pthread_cancel call in main.c:
`sed -i s/pthread_cancel/\/\/pthread_cancel/g' src/main.c`
17. Build nrsc5:
`mkdir build && cd build`
`cmake -DCMAKE_INSTALL_PREFIX=$PREFIX ..`
`make`
`make install`
`cd ../..`

**Here is a video of it running in Android Studio's emulator, connecting to rtl_tcp on my laptop.**

https://github.com/theori-io/nrsc5/assets/139527313/f6a67145-3d79-48f1-aad1-265f920c4703

ferrellsl, 2023-07-27:
Thanks!  I'll build this tomorrow when I get a chance.

ferrellsl, 2023-07-27:
Android audio is based on OpenSL ES so the pulseaudio used by termux is different from the pulseaudio that nrsc5 expects to find.  There's a lot more info regarding pulseaudio and alsa on termux here:

[TRACKER] Integrating ALSA and Pulseaudio into Termux #821
https://github.com/termux/termux-packages/issues/821

drewinchas, 2023-08-01:
So, I've successfully run it on real hardware (Older Galaxy Tab A from 2015) and in the emulator. In both cases, I run it as a regular user (no sudo) and I connect via TCP. I'm able to play/hear audio with both. 

For what it's worth, I get the same USB device error on my real device. My work around (for now) is to use the SDR Driver app (Which includes a TCP server)

ferrellsl, 2023-08-01:
> So, I've successfully run it on real hardware (Older Galaxy Tab A from 2015) and in the emulator. In both cases, I run it as a regular user (no sudo) and I connect via TCP. I'm able to play/hear [...]

Yes, I was aware of that from your earlier posts. I'm wondering if the version of Android (version 13, without root) that I'm using on my Galaxy Tab S7+ has more stringent security requirements.  I'm using Android 12 (rooted) on an Orange Pi 5 .  The Galaxy Tab S7+ will invoke the binary from the command line and display all the available command line arguments, but as soon as I try to stream audio, either to a file or via TCP with the RTL-SDR driver, I get either a permissions error or an unable to open audio device error. 

On the rooted Orange Pi  5 (Android 12), I can run the binary with or without superuser rights and stream audio to a file or via TCP.  I also have a very ancient tablet running LineageOS (Android 7) that I will try later. 

I know I won't get portaudio output on any of my devices until the Termux devs get a usable/compatible version ported to Termux so I'll use TCP or pipes instead for now.  Oddly, I haven't been able to get pipes to work under Termux either.  According to the nrsc5 docs, I should be able to pipe the output like this to mplayer or mpv:  nrsc5 -o - 90.5 0 | mplayer -

Piping to mplayer seems to work but there's no audio ouput, just silence.  When I pipe to mpv like so: nrsc5 -o - 90.5 0 | mpv -
I get the following error:  Failed to recognize file format.

So for now I'll just have to stream via TCP which is no big deal, but ultimately I'd like to be able to obtain audio output in Android in the same manner we do on Windows and Linux.

Using the Android RTL-SDR driver works on my Orange Pi 5 but not on the Galaxy Tab S7+.  When I run the nrsc5 binary on the Galaxy Tab after starting the RTL-TCP server, I get an error saying it's unable to open the audio device. On the Orange Pi 5 it runs, but I get a lot of "lost synchronization" messages and audio pauses that I think are related to the driver and its default settings.   What are the settings you're using in the RTL-SDR driver?  I've only tested the default settings and I think the default buffer size may be too small.

One more thing that folks should note if they're having direct USB access problems, I was only able to achieve that using the version of Termux available thru the F-Droid store.  The version from github and the Google PlayStore would not work at all for me.

ferrellsl, 2023-08-01:
I decided to install a proot Linux on my Orange Pi 5 running Android 13 to see if it would be usable with nrsc5 and a lightweight GUI.  I was pleasantly surprised that it worked and it works well.

Here are the steps I followed, I wasn't documenting this as I went along so these steps are from memory and I may have left out a step or gotten them out of order, but here they are:

1.  Install a VNC viewer and the RTL-SDR driver from the PlayStore.
2.  Install Kali Linux.  I used the Andronix app for that but there are termux scripts out there that will do the same thing.

3.  Inside termux, run this script to get pulseaudio configured in termux for your Linux distro:

pkg install wget && wget https://andronixos.sfo2.cdn.digitaloceanspaces.com/OS-Files/setup-audio.sh && chmod +x
setup-audio.sh && ./setup-audio.sh

4.  Start the pulse audio server while in termux:  pulseaudio --start

5.  Start your vnc server in termux:  vncserver-start

6.  Run the script to start kali:  ./kali-star.sh

7.  Once you've logged into Kali, update everything with a sudo apt update  && sudo apt upgrade

8.  I also installed alsa utils but they may not be necessary.  Next, build nrsc5 per the github instructions.  Kali complained of a missing pkgconfig, so I had to install that as well before a successful build

9.  Install python3

10.  Install a lightweight interface for nrsc5.  I like the one here:  git clone https://github.com/JoeBona1/nrsc5player

11.  Run the the player and configure it for your RTL-TCP server. I just stuck with the default of 127.0.0.1:14423

12.  Select your frequency and press the play button.  It took approx 45 seconds before I got any audio output.

[image]

[image]

ferrellsl, 2023-08-02:
I forgot to mention in the earlier post, if you're using Android 12 or greater and receive an error message about "Process completed (signal 9)" when trying to install a proot Linux distro, then follow the guide here.  It involves enabling developer functions and disabling a "Feature Flag" called "settings_enable_monitor_phantom_procs"

https://docs.andronix.app/android-12/andronix-on-android-12-and-beyond

ferrellsl, 2023-08-02:
Had some extra time on my hands today so I thought I'd see how NRSC5-DUI works by the above method.  It seems to work better than the NRSC5Player that I tried earlier.

[image]

[image]

[image]

drewinchas, 2023-08-02:
That's exciting to see your progress! I'd not heard of Andronix before; that's a cool capability. I was wondering if my versions of Android had any affect on permissions. My Android Emulator is running Android 14 and the physical tablet is running Android 7.1.1

ferrellsl, 2023-08-02:
> That's exciting to see your progress! I'd not heard of Andronix before; that's a cool capability. I was wondering if my versions of Android had any affect on permissions. My Android Emulator is running Android 14 and the physical tablet is running Android 7.1.1

I hadn't heard of Andronix before yesterday either. Several years ago I had played around with several chroot versions of Linux on Android but Google has been continually locking down Android with every release so that most of those distros will no longer work, beginning with Android 12.  I sort of stumbled across Andronix.  It's merely an app that steps you thru the process of picking a proot Linux distro and a desktop manager and then it builds a script that you paste into Termux to download that distro and install it.  I used Andronix to install Kali Linux with an XFCE desktop because it's pretty small and lightweight, but there are several flavors including Ubuntu that you can install along with several desktop managers.

I think the Android emulator has a relaxed security level so as to not impede developers but Android 12 and greater on real hardware are getting so locked down that it gets progressively harder to tinker with.

I was going to apply the same steps above to install proot Kali on my Samsung Galaxy Tab S7+, but the only way to disable the Feature Flag "settings_enable_monitor_phantom_procs" is via an adb shell because Samsung doesn't even allow access to the feature flags under the Developer Options.  I have an Aldocube iPlay 50 Pro running Android 14 and it's the same story there.

I'll see if I can get around to using an adb shell to disable the proper feature flag on my other tablets in a few days.

Here's a link on how to do that yourself  if want to give it a try on your hardware:  

https://www.reddit.com/r/termux/comments/10jen26/how_to_disable_phantom_killer/

ferrellsl, 2023-08-03:
I decided to install Kali on my Alldocube iPlay 50 Pro today.  To disable the phantom process flag, I had to use a USB cable for adb access because apparently wireless adb access is no longer an option on this tablet.

If this is the case with anyone else's hardware, tether your tablet to your PC and open a command prompt where your ADB software is located and type the following after making sure you've enabled USB debugging on the tablet:

adb devices

If everything is working properly you should see something similar to this:
List of devices attached
T1030M128GB23082298     device

Next copy and paste the following 3 lines in the command window.  There won't be any responses if successful:
adb shell "/system/bin/device_config set_sync_disabled_for_tests persistent"
adb shell "/system/bin/device_config put activity_manager max_phantom_processes 2147483647"
adb shell settings put global settings_enable_monitor_phantom_procs false

Restart your tablet and install Termux and the proot Linux distro of your choice.

ferrellsl, 2023-08-03:
The above method for disabling the phantom procs flag was also successful for my Samsung Galaxy Tab S7+ which was surprising being that Samsung can be a real security Nazi.  Here are some pics of the Alldocube iPlay 50 Pro and the Samsung tablet running Kali on Android with nrsc5 doing its thing.  Do yourself a favor if your device doesn't have its own keyboard/mouse/touchpad and get a USB hub and plug in a real keyboard and mouse or you'll lose your mind.

Alldocube with hub.

[image]

[image]

And the Samsung with keyboard and trackpad.

[image]

It would be nice if an Android developer would write a native Android app that takes advantage of the nrsc5 binary built with Termux and the RTL SDR driver so we could avoid having to install Linux on Android in order to have a GUI for nrsc5 on standalone devices.  I'm a Windows programmer and have zero Android programming skills so this will have to do for now.

ferrellsl, 2023-08-09:
I decided to install XFCE under Termux and build a Termux native version of nrsc5-dui and use the Termux native version of nrsc5.  Here's the result.  Sorry for the crappy pic.

[image]

Open Termux and install xfce:  pkg install xfce4

Leave Termux running and install the APK found in the releases section below that matches your environment.
Follow the X11 instructions here:  https://github.com/termux/termux-x11

Once you've Installed the X11 APK from the above repo.  Run the following command in Termux:  termux-x11 :1 -xstartup "dbus-launch --exit-with-session xfce4-session"

You'll see some error messages that you can ignore, now leave Termux running and then run the termux-x11 app and you should be looking at an XFCE desktop.

From there, open a terminal and install the pre-requisites for nrsc5-dui.

Numpy will be an issue as it's not installed via pip.  Install it via:  pkg install python-numpy
Pyaudio will also be a probem.  To install it, you will have to add an additional repo to Termux and run a script.  Run the lines below in your XFCE desktop terminal:

pkg install wget
$PREFIX/bin/wget https://its-pointless.github.io/setup-pointless-repo.sh
bash setup-pointless-repo.sh
pkg install portaudio portaudio-dev
pip3 install pyaudio

If I remember correctly, the other nrsc5-dui dependencies were installed easily using pip or pip3, except for pygobject.

I had to install a couple of other dependencies first:  
pkg install xorgproto
pkg install gobject-introspection
pip install pygobject

pyopensskl complained that I didn't have a Rust complier installed when I tried to pip install it.

Just install Rust first with: pkg install rust
and then install pyopenssl with:  pip install pyopenssl

I copied the nrc5 binary and the libnrsc5.so into the same folder where I cloned nrsc5-dui, ran nrsc5-dui and configured it for RTL-TCP and that was it.

ferrellsl, 2023-08-09:
OK, I finally got nrsc5 running from the command line on non-rooted devices using pipes, mpv and the RTL TCP driver.

Enter the folder where your nrsc5 binary is located and enter the following:

./nrsc5 95.5 0 -H 127.0.0.1:14423 -o - | mpv -

Replace the 95.5 0 with the frequency and channel number of your choice.

ferrellsl, 2024-08-07:
It has been quite a while since I fiddled with this, but it now appears that Android Termux has a version of pulse audio that is compatible with nrsc5, so there's no longer any need for pipes or mpv to play audio from the command line.  Run pulse audio first by typing:  pulseaudio --start.

If you're on  real hardware using a non-rooted device, build nrsc5 within Termux using the following CMAKE arguments to avoid an error when running the final "make install" step:  cmake -DUSE_NEON=ON -DLIBRARY_DEBUG=1 -DBUILD_DOC=ON -DCMAKE_INSTALL_PREFIX:PATH=$PREFIX ..

/usr/local/bin is restricted on non-rooted devices so be sure to use the cmake arguments above to avoid "make install" from erroring out. Then, invoke nrsc5 by typing: nrsc5 your_freq your_channel -H 127.0.0.1:14423  

I still recommend installing Linux or XFCE for Termux to gain access to some python GUIs for nrsc5 if you have the patience for it.