theori-io/nrsc5 #278: Move from FAAD2 to fdk-aac to fix Parametric Stereo implementation?

pclov3r, 2022-06-18:
Was going over this https://github.com/theori-io/nrsc5/issues/260 and discovered that if I checkout this old build https://github.com/theori-io/nrsc5/tree/72c51de9c58f93200d011ade6e0954b27dcc83c4 and load a transcoded ADTS file into audacity it seems play correctly.

If I load that file into VLC it sounds awful and similar to the issues reported here.

Seems like https://github.com/theori-io/nrsc5/issues/88 may also be causing this as well. 

At that time it was proposed to move from FAAD2 to fdk-aac so I'm not sure if that will fix this issue.

Just some obversions I noticed today. 

@argilo not sure if any of this is useful

TheDaChicken, 2022-07-03:
I tried myself by checking out that old build and outputted adts. I played it through FFmpeg and it sounds good.

pclov3r, 2022-07-03:
Was this with with a low bitrate channel that is is in mono with the current build?

TheDaChicken, 2022-07-04:
Yes, it was a low bitrate at 30kbps. I forgot to mention that.

pclov3r, 2022-12-09:
I'll close this in light of https://github.com/theori-io/nrsc5/pull/285

closed 2022-12-09 by pclov3r