theori-io/nrsc5 #279: Long WAV files can cut off

don-code, 2022-06-20:
I recently used the `-o wav-output` feature to capture a five-hour segment of audio. The resulting WAV file cuts off at three hours, 22 minutes. If I import as raw PCM, the full five hours of audio is heard, so all samples have been dumped to the file - the size section just has not been updated. I stopped the capture with Ctrl+C - maybe there is a more appropriate way to stop?

I have not tried another five-hour (or more than 3:22) capture, but could easily do so if that'd be helpful.

Here's the last few lines of output before I stopped the capture:
```
22:58:38 MER: 12.7 dB (lower), 13.6 dB (upper)
22:58:38 BER: 0.000290, avg: 0.000663, min: 0.000016, max: 0.009602
22:58:38 Title: Family Friendly Weekends Start at 6am on Saturday
22:58:38 Artist: Discover 88.9 and WERS.ORG
22:58:38 Audio bit rate: 59.1 kbps
```

Here's the output of `ffprobe` on the output file:
```
Input #0, wav, from 'out.wav':
  Duration: 03:22:53.94, bitrate: 2064 kb/s
  Stream #0:0: Audio: pcm_s16le ([1][0][0][0] / 0x0001), 44100 Hz, stereo, s16, 1411 kb/s
```

Here are the first 100 bytes of the file, showing the header (the full file is 3.1gb):
```
00000000  52 49 46 46 f7 ff ff 7f  57 41 56 45 66 6d 74 20  |RIFF....WAVEfmt |
00000010  28 00 00 00 fe ff 02 00  44 ac 00 00 10 b1 02 00  |(.......D.......|
00000020  04 00 10 00 16 00 10 00  03 00 00 00 01 00 00 00  |................|
00000030  00 00 10 00 80 00 00 aa  00 38 9b 71 64 61 74 61  |.........8.qdata|
00000040  bb ff ff 7f 21 f4 f7 e5  ca f7 7c ec 2f 01 3b f3  |....!.....|./.;.|
00000050  85 0a 5f fb 6c 0d 39 05  9f 08 27 0f 9e 00 c5 16  |.._.l.9...'.....|
00000060  51 fb 58 1b                                       |Q.X.|
```

Here's what I get if I import as "raw PCM" in Audacity with the settings set to 16-bit stereo PCM, just to prove all of the data is indeed dumped and listenable:
[image]

pclov3r, 2022-06-20:
There is a maximum size that is supported for WAV files. AFIK many programs only support 2 GB wav files while others can support up to 4GB. 

I wonder if that's causing the issue here.

Perhaps for very long recordings wav files need to be split?

@argilo any opinions here?

argilo, 2022-06-20:
As @pclov3r mentioned, the maximum size for a WAV file is 4 GB.

The "libao" library, which nrsc5 uses to write out WAV files, writes out a placeholder size of `0x7ffffff7` (slightly less than 2 GB) when it begins creating a WAV file, and updates this to the correct size when `ao_close` is called. However, pressing <kbd>Ctrl+C</kbd> in nrsc5 causes an immediate exit, which skips this shutdown step. To exit cleanly, you can instead press <kbd>Q</kbd>. However, there will still be trouble if the file size exceeds 4 GB.

We could perhaps arrange for the correct size to be written out when <kbd>Ctrl+C</kbd> is pressed by adding an interrupt handler.

TunaLobster, 2022-12-24:
I have also found this to be present for piping the output. My work around that I have been testing has been using timeout and a while loop to restart nrsc5 every 3 hours.

ffmpeg message at 3h22m53.94s:
```[pcm_s16le @ 0x56362c80c940] Multiple frames in a packet.eed=   1x    
[pcm_s16le @ 0x56362c80c940] Invalid PCM packet, data has size 3 but at least a size of 4 was expected
Error while decoding stream #0:0: Invalid data found when processing input
```

Short bash script for automatic restarts and sending to icecast.
```bash
#!/bin/sh
trap 'echo caught interrupt and exiting;exit' INT

while true
do
    time
    timeout --foreground 3h sh -c 'nrsc5 -q -o - 92.5 0 | ffmpeg -i - -vn -f mp3 -c:a libmp3lame [content_type=audio/mp3]icecast://source:PASSWORD@localhost:8000/radio'
    sleep 2
    time
done
```