theori-io/nrsc5 #286: No available formula or cask with the name "/tmp/nrsc5".

fanlessfan, 2022-09-20:
Hello,

I am trying to install on macOS 12.5.1 with Homebrew 3.6.2 and failed.

anyone can help?

thx

> admin@VM4-Homebrew tmp % curl https://raw.githubusercontent.com/theori-io/nrsc5/master/nrsc5.rb > /tmp/nrsc5
>   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
>                                  Dload  Upload   Total   Spent    Left  Speed
> 100   512  100   512    0     0  12516      0 --:--:-- --:--:-- --:--:-- 15058
> admin@VM4-Homebrew tmp % brew install --HEAD -s /tmp/nrsc5
> Warning: No available formula or cask with the name "/tmp/nrsc5".
> Please tap it and then try again: brew tap /tmp
> ==> Searching for similarly named formulae...
> Error: No similarly named formulae found.
> admin@VM4-Homebrew tmp % brew --version
> Homebrew 3.6.2
> Homebrew/homebrew-core (git revision f890e2444f9; last commit 2022-09-20)
> Homebrew/homebrew-cask (git revision bd77d02a44; last commit 2022-09-20)
> admin@VM4-Homebrew tmp % sw_vers
> ProductName:	macOS
> ProductVersion:	12.5.1
> BuildVersion:	21G83

markjfine, 2022-09-20:
Yeah, I recommend you don't use the package from Homebrew, although you should use it for the supporting packages.
Recommend you do a git pull and follow the instructions for building under Linux. Once you have pulled the git directory.
    cd to wherever the nrsc5 root directory is then, 
    $ mkdir build
    $ cd build
    $ cmake [options - see below] ..
    $ make
    $ sudo make install

Make sure you use the appropriate build options in the cmake line that fit your particular architecture (Intel should use SSE, M1 should use NEON):

    -DUSE_NEON=ON            Use NEON instructions. [ARM, default=OFF]
    -DUSE_SSE=ON             Use SSSE3 instructions. [x86, default=OFF]
    -DUSE_FAAD2=ON           AAC decoding with FAAD2. [default=ON]
    -DLIBRARY_DEBUG_LEVEL=1  Debug logging level for libnrsc5. [default=5]
    -DBUILD_DOC=ON           Generate html API documentation [default=OFF]

Once you've done all this, it's easy to just update your local git directory and rebuild as needed.

Let us know how it goes.

fanlessfan, 2022-09-20:
Thanks @markjfine for the suggestions. I compile it follow the unix instruction. and got error below. I guess it should in the package librtlsdr-dev, however brew doesn't have it, so I only installed librtlsdr. any idea to fix it?

```
In file included from /Volumes/tmp/nrsc5/src/decode.c:22:
/Volumes/tmp/nrsc5/src/private.h:4:10: fatal error: 'rtl-sdr.h' file not found
#include <rtl-sdr.h>
         ^~~~~~~~~~~
1 error generated.

brew install librtlsdr-dev
Running `brew update --auto-update`...
==> Auto-updated Homebrew!
Updated 1 tap (homebrew/core).

Warning: No available formula with the name "librtlsdr-dev". Did you mean librtlsdr?
==> Searching for similarly named formulae...
This similarly named formula was found:
librtlsdr ✔
To install it, run:
  brew install librtlsdr ✔
==> Searching for a previously deleted formula (in the last month)...
Error: No previously deleted formula found.
==> Searching taps on GitHub...
Error: No formulae found in taps.
```

markjfine, 2022-09-20:
Interesting. You should have an alias in /usr/local/include/rtl-sdr.h that points to /usr/local/Cellar/librtlsdr/0.6.0/include/rtl-sdr.h, which was installed with the Homebrew. Perhaps it wasn't installed properly. Try removing librtlsdr from Homebrew and reinstalling it.

fanlessfan, 2022-09-20:
Thanks @markjfine . the file does exist, but still has error after I remove and reinstall the librtlsdr.

```
admin@VM4-Homebrew build % cmake -DUSE_SSE=ON  ..
CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE) 
-- Building for x86_64-apple-darwin21.6.0
-- Checking for one of the modules 'fftw3f'
CMake Warning at CMakeLists.txt:53 (message):
  libfftw3f not found.  Building from source.

-- Checking for one of the modules 'librtlsdr'
-- Checking for one of the modules 'libusb-1.0'
-- Checking for one of the modules 'ao'
CMake Warning (dev) at /usr/local/Cellar/cmake/3.24.2/share/cmake/Modules/ExternalProject.cmake:3074 (message):
  The DOWNLOAD_EXTRACT_TIMESTAMP option was not given and policy CMP0135 is
  not set.  The policy's OLD behavior will be used.  When using a URL
  download, the timestamps of extracted files should preferably be that of
  the time of extraction, otherwise code that depends on the extracted
  contents might not be rebuilt if the URL changes.  The OLD behavior
  preserves the timestamps from the archive instead, but this is usually not
  what you want.  Update your project to the NEW behavior or specify the
  DOWNLOAD_EXTRACT_TIMESTAMP option with a value of true to avoid this
  robustness issue.
Call Stack (most recent call first):
  /usr/local/Cellar/cmake/3.24.2/share/cmake/Modules/ExternalProject.cmake:4170 (_ep_add_download_command)
  CMakeLists.txt:104 (ExternalProject_Add)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Build of Doxygen API documentation disabled; use -DBUILD_DOC=ON to enable
-- Configuring done
-- Generating done
-- Build files have been written to: /Volumes/tmp/nrsc5/build
admin@VM4-Homebrew build % make
[ 22%] Built target fftw_external
[ 44%] Built target faad2_external
Consolidate compiler generated dependencies of target nrsc5_object
[ 47%] Building C object src/CMakeFiles/nrsc5_object.dir/decode.c.o
In file included from /Volumes/tmp/nrsc5/src/decode.c:22:
/Volumes/tmp/nrsc5/src/private.h:4:10: fatal error: 'rtl-sdr.h' file not found
#include <rtl-sdr.h>
         ^~~~~~~~~~~
1 error generated.
make[2]: *** [src/CMakeFiles/nrsc5_object.dir/decode.c.o] Error 1
make[1]: *** [src/CMakeFiles/nrsc5_object.dir/all] Error 2
make: *** [all] Error 2

admin@VM4-Homebrew build % ls -l /usr/local/include/rtl-sdr.h
lrwxr-xr-x  1 admin  admin  43 20 Sep 15:43 /usr/local/include/rtl-sdr.h -> ../Cellar/librtlsdr/0.6.0/include/rtl-sdr.h

```

markjfine, 2022-09-20:
If the alias is in usr/local/include, then there must be an issue with your compile environment somewhere. Hopefully you're not installing these Homebrew packages and trying to build this as superuser because that can cause a whole lot of problems.

fanlessfan, 2022-09-20:
this is the first time I use the homebrew and I think install the homebrew require root.

```
admin@VM4-Homebrew ~ % /bin/bash -c "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/HEAD/install.sh)"
==> Checking for `sudo` access (which may request your password)...
Password:
```

markjfine, 2022-09-20:
maybe it's the user and group names 'admin' that's confusing me.

fanlessfan, 2022-09-20:
I removed homebrew and re-install it and got same error. I noticed that the first time I run the cmake I got error about automake is missing and I have to manually install it even through autoconf supposed to install it. Any idea? I just need nrsc5 binary for macOS.

```
cmake -DUSE_SSE=ON  ..
CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE) 
-- The C compiler identification is AppleClang 14.0.0.14000029
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Building for x86_64-apple-darwin21.6.0
CMake Error at CMakeLists.txt:36 (message):
  Missing automake.  Install autoconf package and try again.

-- Configuring incomplete, errors occurred!
See also "/Volumes/tmp/nrsc5/build/CMakeFiles/CMakeOutput.log".
See also "/Volumes/tmp/nrsc5/build/CMakeFiles/CMakeError.log".

admin@VM4-Homebrew bin % brew install automake
==> Downloading https://ghcr.io/v2/homebrew/core/automake/manifests/1.16.5
######################################################################## 100.0%
==> Downloading https://ghcr.io/v2/homebrew/core/automake/blobs/sha256:ae77a247a13ea860236a29b02769f5327395f712413f694d8
==> Downloading from https://pkg-containers.githubusercontent.com/ghcr1/blobs/sha256:ae77a247a13ea860236a29b02769f532739
######################################################################## 100.0%
==> Pouring automake--1.16.5.monterey.bottle.tar.gz
🍺  /usr/local/Cellar/automake/1.16.5: 131 files, 3.5MB
==> Running `brew cleanup automake`...
Disable this behaviour by setting HOMEBREW_NO_INSTALL_CLEANUP.
Hide these hints with HOMEBREW_NO_ENV_HINTS (see `man brew`).
```

fanlessfan, 2022-09-20:
do you know where I can download pre-compiled binary for Mac? thx

markjfine, 2022-09-20:
Think you also needed to install autoconf.

Unfortunately I don't know where to get a pre-compiled version. Macs are configured differently now because of the M1 Macs, so pre-compiled isn't a really good option.

fanlessfan, 2022-09-21:
yes. I installed autoconf, but still has this error.

fanlessfan, 2022-09-21:
I installed both
```
admin@VM4-Homebrew build % brew install automake            
Running `brew update --auto-update`...
==> Auto-updated Homebrew!
Updated 2 taps (homebrew/core and homebrew/cask).

Warning: automake 1.16.5 is already installed and up-to-date.
To reinstall 1.16.5, run:
  brew reinstall automake
admin@VM4-Homebrew build % brew install autoconf
Warning: autoconf 2.71 is already installed and up-to-date.
To reinstall 2.71, run:
  brew reinstall autoconf
```

markjfine, 2022-09-21:
`-- Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE)`

Do you have pkg-config installed?

argilo, 2022-09-21:
It also appears that you do not have `fftw` installed, thereby forcing it to be built from source, which is quite time consuming. I'd suggest installing it with homebrew. As for why `rtl-sdr.h` is not found, could it be that your system has another copy of librtlsdr installed somewhere else that CMake is finding instead of the homebrew version?

fanlessfan, 2022-09-21:
Thank you so much @markjfine  @argilo  I installed both pkg-config and fftw and compiled successfully. 
Another question, when I try to copy the executable file to another Mac and run it, it report error "Library not loaded". the other Mac doesn't have homebrew installed and of course doesn't have those library. Is it possible to link the library statically to one binary file so that it can run anywhere alone without those library?

Thanks again for the help.

argilo, 2022-09-21:
I have not tested this on macOS, but you could perhaps try passing the `-DUSE_SSE=ON` option to CMake and rebuilding.

markjfine, 2022-09-21:
The executable is dependent upon other external dynamic libraries. Unless those libraries are copied as well (and using hte same paths) it won't work.

argilo, 2022-09-21:
Oops, I meant `-DUSE_STATIC=ON`.

fanlessfan, 2022-09-21:
Thanks @argilo . I removed everything from build dir and re-run the `cmake -DUSE_SSE=ON -DUSE_STATIC=ON ..` and make. I got errors below. any idea?

```
[ 85%] Built target nrsc5_object
[ 89%] Linking C shared library libnrsc5.dylib
[ 89%] Built target nrsc5
[ 92%] Linking C static library libnrsc5_static.a
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: libnrsc5_static.a(strndup.c.o) has no symbols
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: libnrsc5_static.a(strndup.c.o) has no symbols
[ 92%] Built target nrsc5_static
[ 96%] Building C object src/CMakeFiles/app.dir/main.c.o
[100%] Linking C executable nrsc5
ld: library not found for -lcrt0.o
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [src/nrsc5] Error 1
make[1]: *** [src/CMakeFiles/app.dir/all] Error 2
make: *** [all] Error 2
```

argilo, 2022-09-21:
I guess that option doesn't work on macOS. (It's undocumented and only tested on Windows.) I'm afraid you'll have to investigate linker options on your own if you'd like to get a static build on macOS.

markjfine, 2022-09-21:
It's looking for the c runtime library for a static build as if this is Linux. I've only seen that needed on Mac-based builds for cross-compiling under mwin64 as a dll for Windows, for obvious reasons. I'm guessing for a static Mac build it's different than for Linux since it uses clang and not c 'proper', so it should really be either aliased or written to find the clang rt.

fanlessfan, 2022-09-21:
Thank you so much @argilo @markjfine. I found below. It seems macOS is special.
```
This option will not work on Mac OS X unless all libraries (including libgcc.a) have also been compiled with -static. Since neither a static version of libSystem.dylib nor crt0.o are provided, this option is not useful to most people.
```

fanlessfan, 2022-09-21:
I copied all the required library to another Mac and can run the nrsc5 binary, but with error below. it can run successfully on my compiling Mac which is VMware VM, so no sound output. any idea?

```
./nrsc5 -H rPi4 99.9 0 
11:49:49 Unable to open audio device.

ls -l /usr/local/lib                      
total 6752
-rwxr-xr-x  1 root  wheel    71024 21 Sep 10:56 libao.4.dylib
-rwxr-xr-x  1 root  wheel  2911104 21 Sep 10:56 libfftw3f.3.dylib
-rwxr-xr-x  1 root  wheel    92656 21 Sep 10:56 librtlsdr.0.dylib
-rwxr-xr-x  1 root  wheel   141136 21 Sep 10:56 libusb-1.0.0.dylib
```

vielmetti, 2022-11-17:
I got a successful build using Homebrew with this:

```
curl https://raw.githubusercontent.com/theori-io/nrsc5/master/nrsc5.rb > /tmp/nrsc5.rb
brew install --HEAD /tmp/nrsc5.rb
```

Not exhaustively tested yet, but everything did build correctly and the binary runs. (Now to dig out my RTL-SDR tuner stick.)

argilo, 2025-03-01:
I'll close this off, since the Homebrew instructions appear to work now. I believe the problem may have been fixed in #294.

closed 2025-03-01 by argilo