theori-io/nrsc5 #293: Error No rule to make target 'faad2-prefix/lib/libfaad_hdc.a',

emrecio, 2022-12-27:
I get through most of the compile process but fail at last step...

[OpenSuSE 15.4...]

╭─erecio@moz /usr/local/src/RTL/nrsc5/foo ‹master●› 
╰─$ make -k                                                                                                            2 ↵
[ 28%] Built target faad2_external
[ 85%] Built target nrsc5_object
[ 89%] Built target nrsc5_static
make[2]: *** No rule to make target 'faad2-prefix/lib/libfaad_hdc.a', needed by 'src/nrsc5'.
make[2]: Target 'src/CMakeFiles/app.dir/build' not remade because of errors.
make[1]: *** [CMakeFiles/Makefile2:133: src/CMakeFiles/app.dir/all] Error 2
make[2]: *** No rule to make target 'faad2-prefix/lib/libfaad_hdc.a', needed by 'src/libnrsc5.so'.
make[2]: Target 'src/CMakeFiles/nrsc5.dir/build' not remade because of errors.
make[1]: *** [CMakeFiles/Makefile2:187: src/CMakeFiles/nrsc5.dir/all] Error 2
make[1]: Target 'all' not remade because of errors.
make: *** [Makefile:136: all] Error 2
make: Target 'default_target' not remade because of errors.

Karlomanio, 2023-12-31:
Is there any workaround for this issue?

whelanh, 2024-11-08:
I have the same error trying to compile on Opensuse Tumbleweed

argilo, 2024-11-08:
I just built nrsc5 in an opensuse/tumbleweed Docker container and didn't have any issue.

Do you have all the dependencies installed? I needed `libao-devel fftw3-devel rtl-sdr-devel libusb-1_0-devel`.

argilo, 2024-11-08:
Here are the commands I used to install dependencies:

```
zypper install -t pattern devel_C_C++
zypper install git cmake libao-devel fftw3-devel rtl-sdr-devel libusb-1_0-devel
```

After that, following the rest of the build instructions in the readme worked fine.

argilo, 2025-03-01:
I'll close this off since I haven't received a response, and building appears to work correctly on openSUSE.

closed 2025-03-01 by argilo

argilo, 2025-03-01:
I have also added openSUSE build instructions to the readme.