theori-io/nrsc5 #295: Feature Request: Dump only DWRO named image files

duracell80, 2023-02-15:
It would be very helpful to reduce the number of writes to an SSD drive when using the --dump-aas-files flag. Excluding station graphics, album art and traffic would really help clean up the directory. The weather radar images are updated roughly every 10 minutes OTA but get overwritten a few times in that time. Same for traffic, if there was a way to filter and dump only the files needed that would really help.

An example of the filename I see for radar:
1569_DWRO_******_rev03_20230215_2200_0621.png

Here I'm dumping the files and sending the audio output to null.

```
$DIR_BASE=/tmp/nrsc5
$DEV=0
$BAND=0
$FREQUENCY=107.5

nrsc5 -o - -d $DEV_IDEX -q --dump-aas-files $DIR_BASE $FREQUENCY $BAND > /dev/null 2>&1 &
```

I can then collect the overlays from the /tmp/nrsc5 directory into my home and into an animation video or gif with ffmpeg.

argilo, 2025-05-07:
Thanks for the suggestion.

I think instead of using filenames, it would be best to determine which data component a file belongs to using the MIME type information in the SIG table. For instance, here's a station that broadcasts weather maps (`TTN_STM_WEATHER = 0xEF042E96`) on port 0820:

```
18:21:12 SIG Service: type=data number=22 name=TTN STM
18:21:12   Data component: id=0 port=0810 service_data_type=66 type=3 mime=FF8422D7
18:21:12   Data component: id=2 port=0820 service_data_type=66 type=3 mime=EF042E96
```

Weather files can then be seen on port 0820:
```
18:21:12 LOT file: port=0810 lot=249 name=TMT_035iuu_1_2_20240612_1514_00f9.png size=5466 mime=4F328CA0 expiry=2024-06-12T15:45:00Z
```

Whether such a feature belongs in nrsc5's sample CLI applications or in external program is another question, but I can already see that it would be more complicated than necessary to implement in an application because the `NRSC5_EVENT_LOT` API event does not include the data component's MIME type (unlike the `NRSC5_EVENT_STREAM` and `NRSC5_EVENT_PACKET` events, which do include it). I think I'll add that into the API so that it's easier to filter LOT files, without having to cross-reference to the SIG table.

To solve your immediate problem of excessive SSD writes, you could perhaps ask nrsc5 to write to a tmpfs mount, and periodically copy out the files you're interested in.

argilo, 2025-05-07:
I made the API enhancement in #407. I'll leave this open until we decide whether it's worth adding LOT filtering capability into the CLI applications.