theori-io/nrsc5 #305: Add support for primary service mode MP2

argilo, 2023-07-04:
As noted in #301, WAMU is using primary service mode MP2, but nrsc5 cannot decode this mode yet.

I've added preliminary support in the [add-mp2 branch](https://github.com/theori-io/nrsc5/compare/add-mp2), and will open a pull request once I've had a chance to do some more testing.

It turns out that WAMU is using the P3 logical channel to send its station logos:

[image]
[image]

My recording doesn't have any album art, but it's possible the P3 logical channel might also be used for that.

/cc @markjfine

argilo, 2023-07-04:
Looking through my collection of recordings, I see that WAMU is not the only station sending out data on the P3 logical channel. WMUZ (Detroit) and WVEZ (Louisville) are also doing so. So these changes will also allow additional files to be decoded from stations using the MP3 mode.

markjfine, 2023-07-04:
I pulled their logo off of their website in nrsc5-dui, which is the same here. Interesting that they'd devote an entire channel for station logos and optional album art.

HD1 is their main audio feed, a simulcast of American University's NPR-affiliated station. The second image is most likely for HD2 named 'Bluegrass Country' [Note: Those look like banjo frets, unless it's a ukulele, but definitely not a guitar 'cause only 4 strings]. Doubt HD1 would use any album art, but HD2 might. Worth playing around to find out for this, as well as other stations in the area that provided no logos via their main channels.

markjfine, 2023-07-04:
Best I can tell WAMU is the only station in this region using P3 for this purpose at the moment. Nothing else looks out of the ordinary other than one station, WKYS, apparently using album art for ads. On another station, WIAD's HD2 (Channel Q) no longer has audio. The multipath being generated by a small plane flying around isn't helping with locking onto several marginal stations, but will continue to check these over time.

argilo, 2023-07-04:
> Interesting that they'd devote an entire channel for station logos and optional album art.

In mode MP2, the P3 logical channel is only 12 kbps. That would be quite low for an audio channel, so offloading data to it seems like a reasonable idea.

> Worth playing around to find out for this, as well as other stations in the area that provided no logos via their main channels.

Using the P3 logical channel for data of any kind seems to be rare. Out of my collection of ~80 recordings, only three stations are doing so.

argilo, 2023-07-04:
WMUZ doesn't have anything in their P3 data channel, it's just an endless stream of empty packets.

WVEZ is sending station logo & traffic data in P3.

argilo, 2023-07-04:
The code is now ready for testing: #306.

PR #306 (2023-07-05): Add support for primary service mode MP2
Fixes #305.

This pull request does the following:

1. Adds support for primary service mode MP2. This mode is very similar to mode MP3, except that the P3 logical channel is only half as wide. Because of this difference, it also uses slightly different interleaver parameters. To accomodate this mode, I've added length arguments to some of the functions that process the P3 logical channel.
2. Adds support for fixed data in logical channels other than P1. To accommodate this, we need one CCC data structure per logical channel, and various functions have  [... cut]
closed 2023-07-05 by argilo