theori-io/nrsc5 #314: On Windows, NRSC5 should quit after reading an I/Q file

pclov3r, 2023-07-07:
On Linux I've noticed NRSC5 quits after reading an I/Q file. 

This currently does not occur on Windows.

argilo, 2023-07-07:
I recently noticed the same issue on Linux when I compiled with `-fsanitize` options turned on. I suspect there might be a race condition somewhere in the shutdown process.

argilo, 2025-04-09:
I expect the problem is that we're using `_getch` to read keyboard input on Windows (vs. `read` elsewhere).

https://github.com/theori-io/nrsc5/blob/1a7fd0e87c8a7e832f27875981396e59fda8c334/src/main.c#L529-L536

After the input file is read, the main thread shuts down the input thread with `pthread_cancel`:

https://github.com/theori-io/nrsc5/blob/1a7fd0e87c8a7e832f27875981396e59fda8c334/src/main.c#L879-L880

The [pthread documentation](https://man7.org/linux/man-pages/man3/pthread_cancel.3.html) states that `read` is a cancelation point, but I expect that `_getch` is not, and therefore `pthread_cancel` has no effect on it.

It seems the input thread jumps through hoops to prevent keystrokes from being displayed, and the code that does that is causing grief.

argilo, 2025-04-09:
The code to disable (and later re-enable) echo on non-Windows platforms is here:

https://github.com/theori-io/nrsc5/blob/1a7fd0e87c8a7e832f27875981396e59fda8c334/src/main.c#L516-L525

The Windows `_getch` function reads a character without echo.

argilo, 2025-04-11:
It looks like `pthread_cancel` does not cancel `read` on Windows, so the switch from `_getch` to `read` in #394 did not solve this issue.

I tried using `select` to wait for input (since it can use a timeout value, or wait for a second input to indicate that the input thread should be stopped) but unfortunately `select` only works for sockets on Windows, and not file descriptors.

I suspect the only way to resolve this problem would be to switch over to Windows-specific function calls like `WaitForSingleObject` and `ReadConsoleInput`.