theori-io/nrsc5 #317: Dependences built when not found are on old versions.

pclov3r, 2023-07-07:
@argilo 

Looking though https://github.com/theori-io/nrsc5/blob/master/CMakeLists.txt I've noticed that several if not all of the dependences pull down old versions. if not found on the system. It appears they all have new releases with various bug fixes. 

Is it worth it to go though and bring these all up to newest versions? It appears there is some bug fixes that might help nrsc5 down the road? Of course they always might bring in new bugs.

argilo, 2023-07-07:
I think keeping the dependencies up to date would be a good idea. Do you have time to do a bit of testing to confirm the current versions work well?

pclov3r, 2023-07-10:
In trying to build a version with deps running into a issue with rtl-sdr https://github.com/osmocom/rtl-sdr/commit/142325a93c6ad70f851f43434acfdf75e12dfe03

Looks like they removed pkgconfig as a requirement for windows https://github.com/osmocom/rtl-sdr/commit/0847e93e0869feab50fd27c7afeb85d78ca04631

Perhaps i'm tired and doing something wrong but I can't get it to find libusb when it comes time to build rtl-sdr. I did update LIBUSB_INCLUDE_DIR to LIBUSB_INCLUDE_DIRS on line 148. 

Likely, I'm doing something very wrong here without realizing. 

Any ideas @argilo :)

pclov3r, 2023-07-10:
Just to add to this it looks like updating faad2 is also a problem. It seems they are working on a new release. Not sure how that might help audio decoding.

Still can't get rtl-sdr to find libusb. For some reason it seems it wants to use the system installed libusb instead of the one being built. 

Looks like this idea will take some work to get everything working correctly.

argilo, 2024-02-16:
rtl-sdr was updated in #336.

argilo, 2024-02-17:
#339 should take care of the rest.

PR #339 (2024-02-17): Update external project versions
Fixes #317.

FFTW and libusb have had various bug fixes since the versions originally used in nrsc5. We may as well pick up the latest.

* https://www.fftw.org/release-notes.html
* https://github.com/libusb/libusb/blob/master/ChangeLog
closed 2024-02-17 by argilo