theori-io/nrsc5 #324: Supplying input IQ at lower sample rate

luarvique, 2023-07-25:
Is there a way to feed NRSC5 with the input IQ stream at a lower sample rate, let us say 400ksps? I am assuming it does not require the entire 1.4Msps input bandwidth to function, does it?

argilo, 2023-07-25:
HD Radio signals have a bandwidth of just under 400 kHz, so the input sample rate must be higher than that. Because of the OFDM spacing, 744187.5 Hz is a convenient sampling rate and nrsc5 uses this internally. 16-bit samples can be fed into the API at this rate:

https://github.com/theori-io/nrsc5/blob/6cbc10649aa3c780407ad18c3560d9ef142f280b/include/nrsc5.h#L602-L611

The Python CLI application also accepts 16-bit samples at this rate (when the `cs16` option is specified):

https://github.com/theori-io/nrsc5/blob/6cbc10649aa3c780407ad18c3560d9ef142f280b/support/cli.py#L38

Because the RTL-SDR only supports sample rates from 225-300 kHz, and > 900 kHz, the standard input rate is 744187.5 Hz * 2 = 1488375 Hz, which is converted to 744187.5 Hz by a half-band decimator.

If you'd like to use another rate, you'd have to do your own sample rate conversion outside nrsc5.

I hope this helps.

closed 2023-07-25 by argilo

luarvique, 2023-07-25:
What if the input is coming from something else than RTL-SDR though? I am trying to connect NRSC5 to the OpenWebRX, for decoding HDRadio broadcasts. Since a lot of OpenWebRX is written in Python, it struggles at high sample rates. It can pipe 400ksps to the NRSC5 standard input though, with the analog FM signal at 0Hz. I guess it can even do 800ksps. 1.48Msps is a bit hard for it.

So, would it be possible to add somewhat arbitrary (selectable from several different choices?) input sampling rate to NRSC5?

argilo, 2023-07-25:
I would suggest using the 744187.5 ksps input available in the API. Supporting other sample rates would require adding a fractional or arbitrary resampler to nrsc5, and I don't have any immediate plans to do that. I suppose it could be useful to support SDRs like Airspy which can only operate at a handful of sample rates.