theori-io/nrsc5 #330: Underruns & cracking noises

TheDaChicken, 2023-10-08:
I am getting: Underruns / cracking noises. 
Here is a video: https://youtu.be/XMswC9g92ZY
I could send anything like IQ recording etc of any type in the process.

Here is my code model:
- Getting SDR samples of complex<floats>
- Resampling to NRSC5's constant SAMPLE_RATE
- Converting to uint8_t for nrsc5.
- Passing it into `nrsc5_pipe_samples_cu8`

This is what I am doing for converting from float to uint8_t:
```c++
void IQConverterFUC::Process(const DataBuffer<std::complex<float>> &input, DataBuffer<uint8_t> &output)
{
    output.SetSize(input.GetSize() * 2);
    for (int i = 0; i < input.GetSize(); ++i)
    {
        auto I = static_cast<uint8_t>((input[i].real() * 128.0f) + 127.4f);
        auto Q = static_cast<uint8_t>((input[i].imag() * 128.0f) + 127.4f);
        output[2 * i] = I;
        output[2 * i + 1] = Q;
    }
}
```

argilo, 2023-10-08:
I would guess that this has the same root cause as #261 (and closely related #177).

argilo, 2023-10-08:
Basically, the library needs to implement proper buffering & delay, and when given input samples it should produce output samples at a fixed ratio. It will be a fair amount of work to get it all right, which is why I haven't taken a stab at it yet.

TheDaChicken, 2023-10-09:
That would be helpful. PortAudio LOVES to grab samples before they are available. Even just exposing some stuff related to buffering could help.

TheDaChicken, 2023-10-13:
Plus another thing I was thinking about: 

Should buffering be implemented in client code or be implemented in the API? The API can just expose stuff related to buffering like latency etc. That way buffering can be managed in any way needed.

argilo, 2023-10-13:
The client code could do some additional buffering of its own if it really needs to, but there will have to be some buffering on the API side to get the latency correct and allow things like analog blending.

TheDaChicken, 2023-10-13:
> The client code could do some additional buffering of its own if it really needs to, but there will have to be some buffering on the API side to get the latency correct and allow things like analog blending.

The reason i asked that question is because my program does the analog "blending" itself since it's not implemented in nrsc5. There is no way for me to tell any latency. Plus, the amount of silence I have to add for no-blending feels kind of weird.

TheDaChicken, 2023-10-13:
I want to make your life easier because you may not have much free time.

TheDaChicken, 2023-11-20:
> I want to make your life easier because you may not have much free time.

Should I try to make my own implementation?

argilo, 2023-11-20:
You're welcome to take a crack at it. I don't have any immediate plans to work on this.

TheDaChicken, 2023-11-24:
I read 1017s.pdf. What I got out of it is `latency * 2 (for unit of 2-per-frame)` gives the length of the buffer before the decoder (so input.c would contain the buffer) due to encoder side. There is a delay in the output of the decoder due to analog based on decoded audio frames.  

I created a simple buffer that just starts outputting frames after latency. I removed the start silence. It cuts out twice or three times at the start. After that, it is already a big improvement apparently. I had a few clickly noises while I was coding this. I didn't hear it anymore. 

I started thinking about audio packets to be more "real-time" Wouldn't it be great for audio frames to be bursty? I am just thinking about PortAudio more than anything.

TheDaChicken, 2023-12-09:
This is definitely out of my area for understanding. I have no clue how to understand whitepapers. I hate to ask. What can you get out of it from a glance?

FoxxMD, 2023-12-18:
@TheDaChicken have you made any more progress? I'm interested in getting this fixed as well...

TheDaChicken, 2023-12-18:
> @TheDaChicken have you made any more progress? I'm interested in getting this fixed as well...

I have made some progress. Still working on it. I made something but it outputs a variable N but it keeps a fixed rate... There are so many things that makes this really hard. For example, being accurate. Analog bending would be based on whatever is buffered from some average rate that was calculated in the buffer.

TheDaChicken, 2024-01-23:
@argilo I am stumped.

I was able to create a regular "jitter buffer" but it accelerates once data arrives 
It doesn't output at a fixed rate but at a (almost) consistent rate of `N` samples.

Is it possible to create a regular jitter buffer but then a consistent rate buffer afterward?

With my limited knowledge of this standard, I don't understand how it's possible to create a fixed delay that would be accelerated.
It doesn't seem like there is a clock source that identifies each Audio PDU? 
The jitter buffer I created determined an average of ~1.4 seconds which sounds about right for 32 * 2048/44100hz

Plus, it would be cool if a buffer was able to be created that properly handles underruns as "not good signal enough" to play.

It makes using NRSC5 API a pain in the butt without some buffer.

TheDaChicken, 2024-04-21:
@argilo

Okay, I am trying again pinging since I have a few ideas about solving the problem and it depends. 

I did a lot of more testing. Delaying output at a fixed ratio would be an anti-pattern because it requires sleeping which is not accurate, etc unless the library is missing another way to grab packets more quickly. Delaying packets based on maximum latency results in packets being average around ~32 which is what we want but it's at a variable rate.

Icecast does a similar thing. It first sends a lot and then slows down  

Here is what I am proposing: 

- `nrsc5_reset_buffer` This resets the delay 
- Only allow one channel at a time through callback to ensure the audio is buffered before playback. 
- A function that passes a pointer to some sort of struct. This struct will store context about the buffer. This allows to grab audio from a blocking/non-blocking function instead of a callback to ensure proper buffering on main.c or users using the library. That also makes it possible to open many channels if somehow playing back 2 channels is something the user wants However, this wouldn't be the greatest idea when adding support for FM I would assume in the library.

PR #399 (2025-05-05): Implement elastic buffering for audio output
Replaces #343.
Fixes #177.
Fixes #261.
Fixes #330.

The NRSC-5 standard uses a variable-rate audio codec. To account for variations in the instantaneous bit rate, the standard specifies that both the transmitter and receiver use an elastic buffer, so that audio packets enter the transmitter and leave the receiver at a fixed rate, even though the number of packets sent in individual audio PDUs may vary. Unfortunately, the standard does not explain how elastic buffering is implemented, so we are left to guess many of the details.

Up until now, nrsc5  [... cut]
closed 2025-05-05 by argilo