theori-io/nrsc5 #332: [Window

Michaelg56, 2023-10-13:
When attempting to run the nrsc5/support/msys2-build I Get the following message..
configure: loading site script /etc/config.site
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... configure: error: unsafe srcdir value: '/C/Users/Michael Gilbert/AppData/Roaming/SPB_Data/nrsc5/build/fftw-prefix/src/fftw_external'
make[2]: *** [CMakeFiles/fftw_external.dir/build.make:92: fftw-prefix/src/fftw_external-stamp/fftw_external-configure] Error 1
make[1]: *** [CMakeFiles/Makefile2:114: CMakeFiles/fftw_external.dir/all] Error 2
make: *** [Makefile:136: all] Error 2

TheDaChicken, 2023-10-17:
The script is meant to be run in the home dir on msys2.
All I can assume is that the path could be way too big. Windows doesn't like long paths.

argilo, 2023-11-06:
It may also be worth checking whether the space in the folder name is causing trouble.

If you're able to get this working, please let me know what the issue was, in case there's something in nrsc5 that can be fixed.

argilo, 2024-02-16:
The same issue was reported again in #338, and the submitter confirmed that building from a path without spaces resolved the issue. We'll have to see whether some fix can be applied to nrsc5 to make it possible to install from any path.