theori-io/nrsc5 #333: Trouble with nrsc5 - Linux

n1moy, 2023-11-19:
Trouble with nrsc5 - Linux 

Have been experimenting with nrsc5 and a RTL-SDR blog V3 dongle. Recently upgraded my aerial which is working well for analog reception.  

Built and running nrsc5 on a Linux Mint Victoria 21.2, Cinnamon 5.8.4 tk, Kernel 6.2.0-36-generic x86_64 machine. [nrsc5 revision 80ab64e]

Decoding the nrsc-5 data is extremely sensitive to gain adjustments both manual and automatic. Not much in feedback from the software relating to gain or signal strength. 

Experiencing the loss of synchronization, odd audio artifacts and drop outs. MER can be all over the map. Typical data follows ...

11:10:54 MER: -5.5 dB (lower), -16.0 dB (upper)
11:10:54 BER: 0.090398, avg: 0.069607, min: 0.041372, max: 0.090398
11:10:54 Lost synchronization

Requesting suggestions resolving these types of problems. 

Thanks in advance,

argilo, 2023-11-19:
nrsc5's automatic gain selection is a bit of a hack, but it should work reasonably well for cases where the antenna is in a fixed position. For best results, you may be better off using an SDR package (Gqrx, SDR#, etc) to find a gain value that gives good SNR without overloading the receiver.

The MER values should give you a reasonably good indication of signal quality. Anything lower than 4-5 dB would indicate a signal that is either too weak or subject to interference.

If you're unable to receive a station that is strong, please make a recording using the `-w <file>` option and share that so we can investigate.

n1moy, 2023-11-24:
Found your suggestion to determine an acceptable gain level using another SDR application was very helpful. This has resolved much of my problem. I aimed for ~ -15 dBFS using gqrx, is this level reasonable ? 

Manual gain adjustment preforms much better than automatic. 

Stations which remain to be a problem are lacking in signal strength and quality. [rectangular sidebars badly dis-formed]  

Thanks for your comments.

argilo, 2023-11-24:
> I aimed for ~ -15 dBFS using gqrx, is this level reasonable ?

Yes, that sounds reasonable. But keep in mind that Gqrx's signal meter only shows the strength of the demodulated signal and not the entire passband of the RTL-SDR. If there are nearby stations that are stronger, you'd want to make sure they are also -15 dBFS or lower.

> rectangular sidebars badly dis-formed

One possible cause is multi-path interference. Moving your antenna (sometimes even just a foot or two) may help.

argilo, 2025-03-02:
FYI, I made an improvement to the automatic gain selection in #370. I hope that it will choose a more appropriate gain value than the old algorithm.

closed 2025-03-07 by argilo