theori-io/nrsc5 #335: Error compiling on WINDOWS

Radiofreq911, 2023-11-24:
-- Build files have been written to: C:/msys64/home/ccrcak1gxl/nrsc5/build/rtlsdr-prefix/src/rtlsdr_
external-build
[ 37%] Performing build step for 'rtlsdr_external'
[  3%] Building C object src/CMakeFiles/rtlsdr_shared.dir/librtlsdr.c.obj
make[5]: *** [src/CMakeFiles/rtlsdr_shared.dir/build.make:76: src/CMakeFiles/rtlsdr_shared.dir/librtlsdr.c.obj] Error 1
make[4]: *** [CMakeFiles/Makefile2:163: src/CMakeFiles/rtlsdr_shared.dir/all] Error 2
make[3]: *** [Makefile:136: all] Error 2
make[2]: *** [CMakeFiles/rtlsdr_external.dir/build.make:86: rtlsdr-prefix/src/rtlsdr_external-stamp/rtlsdr_external-build] Error 2
make[1]: *** [CMakeFiles/Makefile2:166: CMakeFiles/rtlsdr_external.dir/all] Error 2
make: *** [Makefile:136: all] Error 2

Radiofreq911, 2023-11-24:
Here is a larger snip of the compile output

[ 27%] Completed 'libusb_external'
[ 27%] Built target libusb_external
[ 29%] Creating directories for 'rtlsdr_external'
[ 30%] Performing download step (git clone) for 'rtlsdr_external'
Cloning into 'rtlsdr_external'...
HEAD is now at f68bb2f lib: Add workaround for Linux usbfs mmap() bug
[ 32%] No update step for 'rtlsdr_external'
[ 33%] No patch step for 'rtlsdr_external'
[ 35%] Performing configure step for 'rtlsdr_external'
CMake Deprecation Warning at CMakeLists.txt:24 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Build type not specified: defaulting to release.
-- Extracting version information from git describe...
-- Udev rules not being installed, install them with -DINSTALL_UDEV_RULES=ON
-- Building with kernel driver detaching disabled, use -DDETACH_KERNEL_DRIVER=ON to enable
-- Building for version: v0.6.0-2-gf68b / 0.6git
-- Using install prefix: C:/msys64/home/ccrcak1gxl/nrsc5/build/rtlsdr-prefix
-- Configuring done (2.9s)
-- Generating done (0.2s)
-- Build files have been written to: C:/msys64/home/ccrcak1gxl/nrsc5/build/rtlsdr-prefix/src/rtlsdr_
external-build
[ 37%] Performing build step for 'rtlsdr_external'
[  3%] Building C object src/CMakeFiles/rtlsdr_shared.dir/librtlsdr.c.obj
make[5]: *** [src/CMakeFiles/rtlsdr_shared.dir/build.make:76: src/CMakeFiles/rtlsdr_shared.dir/librtlsdr.c.obj] Error 1
make[4]: *** [CMakeFiles/Makefile2:163: src/CMakeFiles/rtlsdr_shared.dir/all] Error 2
make[3]: *** [Makefile:136: all] Error 2
make[2]: *** [CMakeFiles/rtlsdr_external.dir/build.make:86: rtlsdr-prefix/src/rtlsdr_external-stamp/rtlsdr_external-build] Error 2
make[1]: *** [CMakeFiles/Makefile2:166: CMakeFiles/rtlsdr_external.dir/all] Error 2
make: *** [Makefile:136: all] Error 2

argilo, 2023-12-02:
Which MSYS2 environment did you use?

I built nrsc5 in the MSYS2 MINGW64 environment just now, and it worked fine.

Radiofreq911, 2023-12-02:
I tried mingw32 and 64 on windows

GRANT LEWIS

Senior Support Engineer – Tiger Team

Centralized Engineering

iHeartMedia | Technology Operations

America’s #1 Audio Company

Reaching 9 out of 10 Americans Every Month

Radio | Digital | Social | Podcasts | Influencers | Data | Events

CAUTION: This message originated from outside of the company. Please take all precautions when opening attachments or links from outside sources.

argilo, 2023-12-02:
OK. Unfortunately I can't reproduce the problem, so you'll have to investigate on your own.

argilo, 2024-02-16:
Are you still having this problem? We've updated to a more recent version of the rtlsdr package, so things may be different now.

argilo, 2024-03-22:
I haven't heard anything back on this, so I'll close off the issue.

closed 2024-03-22 by argilo