theori-io/nrsc5 #338: [Windows] Error Compiling

kllkradio, 2024-02-15:
I am experiencing a similar problem to that of issue #335.   I tried two different computers and got the same error message.  One computer is WIN10, 22H2 and the other is WIN11, 23H2.  MSYS2 was a fresh install in both computers.  I followed the steps specified in the NRSC5 readme exactly.  

I used mingw32.exe to compile the program.

Here is the error message:

checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... configure: error: unsafe srcdir value: '/C/msys64/home/Brian J. Henry/nrsc5/build/fftw-prefix/src/fftw_external'
make[2]: *** [CMakeFiles/fftw_external.dir/build.make:92: fftw-prefix/src/fftw_external-stamp/fftw_external-configure] Error 1
make[1]: *** [CMakeFiles/Makefile2:114: CMakeFiles/fftw_external.dir/all] Error 2
make: *** [Makefile:136: all] Error 2

argilo, 2024-02-15:
I suspect the problem is that spaces in the path name are not handled correctly. You could try building from a location that does not have any spaces to see if that helps.

kllkradio, 2024-02-16:
Thanks for the suggestion.  I used another computer that didn't put spaces in the path name and I was then able to compile the program.  I ran the test file and it worked correctly.  I then used the command line to address the RTL-SDR and successfully decoded an off the air HD signal!

closed 2024-02-16 by kllkradio

argilo, 2024-02-16:
Thanks for confirming my suspicion about the source of the problem. Hopefully a fix can be found so that compilation will succeed regardless of the path.