theori-io/nrsc5 #340: Updating MacPorts port for nrsc5

trodemaster, 2024-03-28:
Hi, I'm the maintainer for the unofficial macports port of nrsc5. One thing I would like to update is the build process downloading fftw instead of picking up the version already installed. Is there a search path I need to configure at build time? 

https://trac.macports.org/ticket/67860

Thanks for any guidance here!

argilo, 2024-03-28:
I haven't looked at portfiles before, but it looks to me like fftw is not listed as a build dependency. Is that the problem?

trodemaster, 2024-03-31:
I added fftw as a build dependency, but it's still not being found at build time. The logic I need to focus on is in the CMakeLists.txt file. macports has it's own structure under /opt that I need to have cmake look in.

trodemaster, 2024-03-31:
Ah there was another "sub port" that was needed to provide fftw 3 single precision library. https://github.com/macports/macports-ports/pull/23301/files

FYI here is the port page
https://ports.macports.org/port/nrsc5/details/

Closing this as all that remains is to work the process to get my PR merged. Thanks for nrsc5 I listen to a lot of HD radio using this program.

closed 2024-03-31 by trodemaster

argilo, 2024-03-31:
I'm glad you were able to solve it.

I suspect you can remove the `faad2` dependencies, since nrsc5 must build its own patched version of faad2 which adds HDC support.