theori-io/nrsc5 #341: The Story of the FM IBOC MP11 Mode

markjfine, 2024-03-29:
Not an issue, but thought it provided some interesting info coming from an iHeart Sr Ops Engr:
The Story of the FM IBOC MP11 Mode[https://www.radioworld.com/tech-and-gear/digital-radio/the-story-of-the-fm-iboc-mp11-mode](url)

argilo, 2024-03-29:
Interesting! I'll have a read over that.

I also recently wrote a blog post about my earlier experiment with MP11: https://irrational.net/2024/03/25/hacking-a-sangean-hdt-20/

argilo, 2024-03-29:
Is there anyone in the Dayton area who could make a recording of WTUE?

argilo, 2024-03-30:
Someone was kind enough to send me a recording of WTUE, and it appears to be doing MP11 according to the standard. My hacked Sangean HDT-20 was able to receive the P4 logical channel, and I was also able to receive it with nrsc5 after removing the workaround that I put in for WWWT (d96b72c304f1902b4a762b632c8f56da6675f64a).

To me this suggests that something is wrong with WWWT's signal. I wonder if that's still the case today.

markjfine, 2024-03-31:
I'll try to get a new recording of WWWT today.

argilo, 2024-04-02:
I'll close this issue off, since #342 made it possible to receive WTUE's P4 logical channel.

closed 2024-04-02 by argilo

ferrellsl, 2024-04-02:
> Not an issue, but thought it provided some interesting info coming from an iHeart Sr Ops Engr: [The Story of the FM IBOC MP11 Mode](https://www.radioworld.com/tech-and-gear/digital-radio/the-story-of-the-fm-iboc-mp11-mode)

This was a very interesting read and and an intriguing way to increase channels and bitrate.  Maybe I missed it during reading, but are current HD receivers capable of decoding MP11?  To me it seemed implied that newer receivers would be required or older receivers would need firmware updates to access these additional channels.

argilo, 2024-04-02:
As far as I know, receiver chips have had MP11 capability for a long time, but for reasons unknown it was disabled by default starting around 2014 (when the `HD_SERVICE_MODE_CONTROL_MP11_ENABLE` property was added to the SI468x firmware). I have yet to encounter a radio that supports MP11, so I expect that very few have overridden that default.

Radios made before 2014 may well be able to receive MP11, and that could explain why the 2014 Toyota Tundra mentioned in the article was able to receive it.

ferrellsl, 2024-04-02:
> As far as I know, receiver chips have had MP11 capability for a long time, but for reasons unknown it was disabled by default starting around 2014 (when the `HD_SERVICE_MODE_CONTROL_MP11_ENABLE` [...]

That's too bad.  I was hoping broadcasters in the Phoenix area would start using MP11 because I would like some higher bitrate channels for music on my car's infotainment system, but it's doubtful they'll ever do so based on your knowledge of current receivers.  Chicken and the egg sort of problem.  I have some old and very rare Insignia NS-HD02 and NS-HD01 receivers that would probably work if any broadcasters in Phoenix start playing with MP11.

argilo, 2024-04-02:
Given that so many receivers have MP11 disabled, I doubt broadcasters will put anything important there for some time. WTUE's HD4 has the same content as its HD2, and WWWT's HD5 is only there so that another station could shut down its AM transmitter and remain within FCC rules with its FM translator.

argilo, 2024-04-02:
The only other options to add more digital capacity are:

* Switch to one of the new "Advanced" Primary Service Modes.
* Turn off analog FM and go all-digital.

These options were only standardized in NRSC-5-E (published in January 2023) and I'm not aware of any receivers that support them yet. So unfortunately the chicken & egg problem will be even larger there.