theori-io/nrsc5 #345: Error: Passed `nil` into T.must

w1ebr, 2024-05-17:
This error occurs when building via Homebrew:

Error: Passed `nil` into T.must
/opt/homebrew/Library/Homebrew/vendor/bundle/ruby/3.3.0/gems/sorbet-runtime-0.5.11380/lib/types/_types.rb:222:in `must'
/opt/homebrew/Library/Homebrew/formula.rb:2582:in `bottle_hash'
/opt/homebrew/Library/Homebrew/sbom.rb:37:in `create'
/opt/homebrew/Library/Homebrew/vendor/bundle/ruby/3.3.0/gems/sorbet-runtime-0.5.11380/lib/types/private/methods/call_validation_2_7.rb:473:in `bind_call'
/opt/homebrew/Library/Homebrew/vendor/bundle/ruby/3.3.0/gems/sorbet-runtime-0.5.11380/lib/types/private/methods/call_validation_2_7.rb:473:in `block in create_validator_method_skip_return_fast2'
/opt/homebrew/Library/Homebrew/formula_installer.rb:834:in `finish'
/opt/homebrew/Library/Homebrew/vendor/bundle/ruby/3.3.0/gems/sorbet-runtime-0.5.11380/lib/types/private/methods/call_validation_2_7.rb:652:in `bind_call'
/opt/homebrew/Library/Homebrew/vendor/bundle/ruby/3.3.0/gems/sorbet-runtime-0.5.11380/lib/types/private/methods/call_validation_2_7.rb:652:in `block in create_validator_procedure_fast0'
/opt/homebrew/Library/Homebrew/upgrade.rb:238:in `install_formula'
/opt/homebrew/Library/Homebrew/install.rb:350:in `install_formula'
/opt/homebrew/Library/Homebrew/install.rb:301:in `block in install_formulae'
/opt/homebrew/Library/Homebrew/install.rb:300:in `each'
/opt/homebrew/Library/Homebrew/install.rb:300:in `install_formulae'
/opt/homebrew/Library/Homebrew/cmd/install.rb:297:in `run'
/opt/homebrew/Library/Homebrew/vendor/bundle/ruby/3.3.0/gems/sorbet-runtime-0.5.11380/lib/types/private/methods/call_validation.rb:270:in `bind_call'
/opt/homebrew/Library/Homebrew/vendor/bundle/ruby/3.3.0/gems/sorbet-runtime-0.5.11380/lib/types/private/methods/call_validation.rb:270:in `validate_call'
/opt/homebrew/Library/Homebrew/vendor/bundle/ruby/3.3.0/gems/sorbet-runtime-0.5.11380/lib/types/private/methods/_methods.rb:277:in `block in _on_method_added'
/opt/homebrew/Library/Homebrew/brew.rb:92:in `<main>'
Please report this issue:
  https://docs.brew.sh/Troubleshooting

argilo, 2024-05-22:
Thanks for the report.

I tried installing nrsc5 through Homebrew here, and it works fine. From the stack trace you provided, it appears the problem lies in your Homebrew installation rather than in the nrsc5 formula. If you find otherwise, then feel free to re-open this issue.

closed 2024-05-22 by argilo

w1ebr, 2024-05-23:
Thank you! I removed and reinstalled and it worked. I also changed the CMake source so that I could specify -DUSE_NEON=ON and it would build for arm64.

argilo, 2024-05-23:
I'm glad to hear it.