theori-io/nrsc5 #346: API Improvements: Document Requirements / Add Missing Constants

TheDaChicken, 2024-06-20:
In the API, 

`nrsc5_pipe_samples_cs16` assumes input went through the half-band filter. This isn't documented as a requirement/function / is not labeled to require in the function name. 

I encountered a problem using 16 bits as a constant type for piping data from different SDRs which required the half-band filter. I have to implement a half-band filter to pipe that type.  

API does not include a few constants such as which may be required:
- IQ sample rate 
- audio sample rate
- audio sample count
- audio channel count (not very important)

Silly small tidbit: 
``__declspec(dllexport)`` is meant to be used in the header file.

argilo, 2024-06-20:
> `nrsc5_pipe_samples_cs16` assumes input went through the half-band filter.

Actually there is no requirement to use a half-band filter. The only requirement is that the input sample rate is 744187.5. I agree that it would be useful to add the required sample rates to the API documentation (1488375 for `nrsc5_pipe_samples_cu8`, 744187.5 for `nrsc5_pipe_samples_cs16`).

We could also document that the audio sample rate is always 44100.

argilo, 2024-06-20:
The reason for 744187.5 is that it is the most convenient sample rate for OFDM demodulation. Unfortunately, the RTL-SDR does not natively support this sample rate (it allows only 225000-300000 and 900000-3200000). Because of this limitation, nrsc5 samples at 744187.5 * 2 = 1488375 and uses a decimating half-band filter to reach the desired sample rate. The API allows samples to be injected either before or after nrsc5's filter. If samples are injected after the filter, the application can use whatever means it likes to achieve the correct sample rate.

TheDaChicken, 2024-06-20:
> Actually there is no requirement to use a half-band filter. The only requirement is that the input sample rate is 744187.5. I agree that it would be useful to add the required sample rates to the API documentation (1488375 for `nrsc5_pipe_samples_cu8`, 744187.5 for `nrsc5_pipe_samples_cs16`).

Thanks for letting me know. *my oblivious nature is a good example that it should probably be documented* 

> We could also document that the audio sample rate is always 44100.

Would be very useful. I have constants I added in my code that would make sense to be added.

TheDaChicken, 2024-06-22:
Should i make pull request for this?

argilo, 2024-06-24:
Yes, that would be useful.

pclov3r, 2024-06-25:
On the topic, It would be nice to see CS16 supported in NRSC5 for I/Q input without use the of API.  https://github.com/theori-io/nrsc5/issues/284

Also, Maybe some GNuradio flowgraphs for both cu8 and cs16 samples would be good to include?

TheDaChicken, 2024-06-25:
> I find myself taking I/Q files from say SDRconsole and running though GNUradio flowgraph to get a usable CU8 file for use in NRSC5. If I could keep those 16-bit I figure it should yield better performance.

A quick look, it looks like that would require another function in the API. For example, `nrsc5_open_file_cs16`
Would that be a good idea?

pclov3r, 2024-06-25:
> > I find myself taking I/Q files from say SDRconsole and running though GNUradio flowgraph to get a usable CU8 file for use in NRSC5. If I could keep those 16-bit I figure it should yield better performance.
> 
> A quick look, it looks like that would require another function in the API. For example something like: `nrsc5_open_file_cs16` OR switch to loading the file in main.c? Would that be a good idea?

That would have to be @argilo call to make :) 

Trying to create a gnuradio flowgraph now for cs16.

pclov3r, 2024-06-25:
I can't get CS16 input working with the the current Python cli in a file converted in GNUradio. Works perfectly fine for cu8 with flow graph over at https://github.com/theori-io/nrsc5/issues/33#issuecomment-668352103

I am feeding it 744187.5 resampled from GNUradio. Likely something I'm doing wrong.

Please excuse my poor DSP knowledge :)

[image]

pclov3r, 2024-06-28:
Your probably busy @argilo but when you get chance to answer the question form @TheDaChicken and my issue that would be awesome :)

pclov3r, 2024-07-19:
@argilo If your back from your vacation and can look at the above that would be awesome :) 

@TheDaChicken Not sure if your done with https://github.com/theori-io/nrsc5/pull/347 yet

TheDaChicken, 2024-07-19:
> @TheDaChicken Not sure if your done with #347 yet

I'd say it is done. A new pull request can be done for #284

TheDaChicken, 2024-09-01:
The constants I wanted to document have been added. We can continue the IQ 16 bits discussion in #284

closed 2024-09-01 by TheDaChicken