theori-io/nrsc5 #352: COMM Tag

TheDaChicken, 2024-08-18:
I wanted to post that I found a station over here (97.3 KIRO) that is using the COMM tag. 

[image]

I don't know how useful the comment tag is. Seems it's just a simple Language field, a short description field, and a content field providing a description.

markjfine, 2024-08-18:
Looks like embedded URLs for links to ads.
First two internet company: `eng+omm https://www.quantumfiber.com/`
Last long one is a heat pump company: `eng+omm https://homecomfortalliance.com/?keyword=home%20comfort%20alliance&matchtype=e&utm-source=google&utm-medium=cpc&utm-campaign=168

TheDaChicken, 2024-08-19:
> Looks like embedded URLs for links to ads. First two internet company: `eng+omm https://www.quantumfiber.com/` Last long one is a heat pump company: `eng+omm https://homecomfortalliance.com/?keyword=home%20comfort%20alliance&matchtype=e&utm-source=google&utm-medium=cpc&utm-campaign=168`
> 
> Leading null is a little odd.

Very interesting! Kind of makes sense but also doesn't. Sounds like they could have used a Commerical tag 🤷

markjfine, 2024-08-19:
I've kinda lost the bubble on the internal format of all this. Wouldn't surprise me if that's exactly what they thought it was for. After seeing how WWWT took liberties with implementing a fifth subchannel, it also wouldn't surprise me if there's a lot of experimentation or flat-out rougue implementation going on while there's no defacto FCC oversight.

TheDaChicken, 2024-08-19:
> I've kinda lost the bubble on the internal format of all this. Wouldn't surprise me if that's exactly what they thought it was for. 

Yep. Kind of sounds like it. 

> After seeing how WWWT took liberties with implementing a fifth subchannel, it also wouldn't surprise me if there's a lot of experimentation or flat-out rougue implementation going on while there's no defacto FCC oversight.

I saw you post that. I thought that was interesting but pretty frustrating. I mean, it could be worse. 😭

argilo, 2024-08-31:
Interesting. It looks like COMM is the only tag defined in the NRSC-5 standard that nrsc5 doesn't explicitly decode. I might not have added support since I hadn't seen it used yet.

[image]

The format is described here: https://id3.org/id3v2.3.0#Comments

```
<Header for 'Comment', ID: "COMM">
Text encoding           $xx
Language                $xx xx xx
Short content descrip.  <text string according to encoding> $00 (00)
The actual text         <full text string according to encoding>
```

argilo, 2024-08-31:
@TheDaChicken If you can share an I/Q recording or two, that would be helpful for testing.

TheDaChicken, 2024-08-31:
> Interesting. It looks like COMM is the only tag defined in the NRSC-5 standard that nrsc5 doesn't explicitly decode. I might not have added support since I hadn't seen it used yet.

I noticed `COMR` is decoded in nrsc5. Is not publically useable (only `log_debug`). That's something that can be worked on as well. Though, I haven't seen `COMR` used myself. 

> @TheDaChicken If you can share an I/Q recording or two, that would be helpful for testing.

Sure. I will be able on Tuesday night - Am currently hanging out with my grandparents for Labor day weekend in a different city.

argilo, 2024-08-31:
> I noticed COMR is decoded in nrsc5. Is not publically useable (only log_debug).

The reason I didn't add it to the API is that I have never seen anything other than placeholder information in a COMR tag. If stations start using it for real, then it might become a useful addition.

TheDaChicken, 2024-09-01:
> The reason I didn't add it to the API is that I have never seen anything other than placeholder information in a COMR tag. If stations start using it for real, then it might become a useful addition.

Yep, completely makes sense + no way of testing it too since no stations use it.

argilo, 2024-09-01:
There are actually quite a few stations that send out COMR tags, so it's possible to test. But in every case the payload is the same useless placeholder data.

TheDaChicken, 2024-09-01:
> There are actually quite a few stations that send out COMR tags, so it's possible to test. But in every case the payload is the same useless placeholder data.

Ah, okay. Weird. I wish stations used it.

argilo, 2024-09-01:
Here are all the COMR samples I have from my recordings:

CITE: price=CAD000.00 until=2025-05-20 url="-" seller="-" desc="-" received_as=0
CJMJ: price=CAD000.00 until=2025-06-12 url="-" seller="-" desc="-" received_as=0
KUNV: price=$0.00 until=2099-12-31 url="www.url.com" seller="Seller Name" desc="Product Description" received_as=0
WHRO: price=USD000.00 until=2025-07-27 url="-" seller="-" desc="-" received_as=0
WHRV: price=USD000.00 until=2025-07-27 url="-" seller="-" desc="-" received_as=0
WPOZ: price=USD000.00 until=2025-08-26 url="-" seller="-" desc="-" received_as=0

markjfine, 2024-09-01:
Looks like random radio station call letters.

argilo, 2024-09-01:
The call signs are the stations I heard each value from.

TheDaChicken, 2024-09-04:
> @TheDaChicken If you can share an I/Q recording or two, that would be helpful for testing.

Here are some I/Q recordings:
https://www.dropbox.com/scl/fi/p5ccdrhj9iwgr3eqmamf5/KIRO-9-3-24-703PM.raw?rlkey=jp6wase9ofhtrz0fxsux72i9w&st=6tylbmxk&dl=1 (small)
https://www.dropbox.com/scl/fi/vzsvzikmit8i8c51nuaec/KIRO-9-3-24-732PM.raw?rlkey=5ng11mxe6ukq3brsap161n5yw&st=mjv15nec&dl=1 (large)

argilo, 2024-09-04:
Got them, thanks!

PR #375 (2025-03-20): Implement COMM Tag
Fixes #352 

KIRO-FM here in Seattle uses the only unimplemented id3 tag in nrsc5. 

This tag in use:
```
20:52:03 BER: 0.000000, avg: 0.000000, min: 0.000000, max: 0.000000
20:52:04 Audio bit rate: 19.3 kbps
20:52:05 MER: 13.9 dB (lower), 14.0 dB (upper)
20:52:05 BER: 0.000000, avg: 0.000000, min: 0.000000, max: 0.000000
20:52:05 Title: Seattle Sports
20:52:05 Artist: Brought to you by Quantum Fiber
20:52:05 Album: Seattle Sports
20:52:05 Genre: Sports
20:52:05 Comment: lang=eng Comm https://www.quantumfiber.com
20:52:06 MER: 13.9 dB (lower), 14.0 dB (upper)
20:52:06 BER: 0 [... cut]
closed 2025-03-20 by argilo