theori-io/nrsc5 #355: Frequently lose synchronization

psteckler, 2025-01-26:
For some FM stations here in the Bay Area, `nrsc5` works just fine for me, with clear audio output. For other stations, either I frequently lose synchronization, or the sound is very choppy.

Are these problems likely due to poor signal strength? A poor SDR dongle?

Example logging output:
```
$ nrsc5 -l 1  -g 50.0 97.3 2 
Found Rafael Micro R820T tuner
Exact sample rate is: 1488375.071248 Hz
[R82XX] PLL not locked!
Allocating 8 zero-copy buffers
23:12:41 Synchronized
23:12:42 MER: -10.8 dB (lower), -6.5 dB (upper)
23:12:42 BER: 0.205262, avg: 0.205262, min: 0.205262, max: 0.205262
23:12:42 Lost synchronization
23:12:44 Synchronized
23:12:45 MER: -10.2 dB (lower), 2.5 dB (upper)
23:12:45 BER: 0.253100, avg: 0.229181, min: 0.205262, max: 0.253100
23:12:45 Lost synchronization
23:12:47 Synchronized
23:12:48 MER: -9.5 dB (lower), 0.2 dB (upper)
23:12:48 BER: 0.260078, avg: 0.239480, min: 0.205262, max: 0.260078
23:12:48 Lost synchronization
23:12:50 Synchronized
23:12:51 MER: -12.1 dB (lower), 2.1 dB (upper)
23:12:51 BER: 0.252301, avg: 0.242686, min: 0.205262, max: 0.260078
23:12:51 Lost synchronization
23:12:53 Synchronized
```

markjfine, 2025-01-26:
If you look at the MER, it looks like the lower data sideband is getting crushed, likely from an adjacent station. nrsc5 is probably trying to synch with the stronger, upper one, which is neither all that strong nor stable.

Just to verify: You may want to look at what the signal looks like for standard FM using a regular tuning tool like SDR# or GQRX. If you can barely see the two pillars on either side of the signal then it's either not strong enough and/or being interfered with.

That all said, have you tried not setting the gain? I've found that setting the gain too high is a double-edged sword. You're amplifying the desired signal, but you're also amplifying the interference, as well as possibly overdriving the front end. Try leaving the '-g 50.0' part out and see how it performs without it.

psteckler, 2025-01-26:
Thanks, I'll try these suggestions.

closed 2025-01-26 by psteckler