theori-io/nrsc5 #359: Improved multipath and fading performance

pclov3r, 2025-02-19:
Talking with Argilo they believe that nrsc5 can handle fading and multipath much better. Especially in a moving vehicle.

pclov3r, 2025-02-19:
If necessary, I have a bunch of IQ recordings of less then ideal channel conditions. I could post them.

argilo, 2025-02-19:
Yeah, the current OFDM equalization algorithm was built under the assumption of a fixed receiving antenna, which will not see much change in the channel state from one OFDM symbol to the next. (Symbols are ~2.9 milliseconds long.) The equalizer seems to perform badly when the receiving antenna is moving, for instance in a car. Synchronization is frequently lost.

Probably the equalization should be performed independently for each OFDM symbol. Or the fraction of state which is carried over from one symbol to the next should be greatly reduced.

> If necessary, I have a bunch of IQ recordings of less then ideal channel conditions. I could post them.

I'm fine for the moment. I have a collection of recordings that I made from a moving vehicle.

I too could post some, if anyone else wants to take a crack at improving the equalizer.