theori-io/nrsc5 #361: Using the Python API CTRL+C or Q does not work to quit the script when reading I/Q files.

pclov3r, 2025-02-21:
Something I noticed is when using the Python API CTRL+C or Q does not work to quit the script when reading I/Q files.

When trying Ctrl+C i get

`^CException ignored on calling ctypes callback function: <function NRSC5._set_callback.<locals>.callback_closure at 0x7f2ddf236ac0>
Traceback (most recent call last):
  File "/usr/local/src/nrsc5/support/nrsc5.py", line 606, in callback_closure
    self._callback_wrapper(evt)
  File "/usr/local/src/nrsc5/support/nrsc5.py", line 510, in _callback_wrapper
    self.callback(evt_type, evt)
  File "/usr/local/src/nrsc5/support/cli.py", line 18, in <lambda>
    self.radio = nrsc5.NRSC5(lambda evt_type, evt: self.callback(evt_type, evt))
                                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/src/nrsc5/support/cli.py", line 206, in callback
    self.audio_queue.put(evt.data)
  File "/usr/lib/python3.11/queue.py", line 140, in put
    self.not_full.wait()
  File "/usr/lib/python3.11/threading.py", line 320, in wait
    waiter.acquire()
KeyboardInterrupt: `

Q doesn't seem to do anything?

argilo, 2025-02-21:
I have noticed this too (and been annoyed by it). The script is multi-threaded and will have to handle cleanup more carefully when a `KeyboardInterrupt` occurs.

argilo, 2025-02-21:
Actually, it looks like thread cleanup is handled properly. The real problem appears to be that `KeyboardInterrupt` can occur inside the callback function, which does not handle exceptions. I don't think Python exceptions can pass through with is ultimately a C callback, so I suppose the Python callback function will have to catch `KeyboardInterrupt` and set a state variable to let the main loop know that it should stop.

argilo, 2025-02-25:
Some notes on Ctrl+C handling:

When Ctrl+C is pressed, the exception often occurs inside a ctypes callback, which is set up here:

https://github.com/theori-io/nrsc5/blob/2fa179b85b5a9d13d60fdc9ffee22e7914a59ef1/support/nrsc5.py#L606-L611

When an exception occurs inside the callback, ctypes ignores it:

```
Exception ignored on calling ctypes callback function: <function NRSC5._set_callback.<locals>.callback_closure at 0x72edf84e91c0>
Traceback (most recent call last):
  File "/home/argilo/git/nrsc5/support/nrsc5.py", line 608, in callback_closure
    def callback_closure(evt, opaque):

KeyboardInterrupt:
```

This is by design, since a C program has no way to handle a Python exception.

I tried adding a `try` / `catch` block around the `self._callback_wrapper(evt)` line, but even that isn't enough. The `KeyboardInterrupt` still often occurs outside the `try` block and is therefore ignored.

Registering a signal handler to catch `SIGINT` seems to work:

```diff
diff --git a/support/cli.py b/support/cli.py
index d6a177e..5e701e7 100755
--- a/support/cli.py
+++ b/support/cli.py
@@ -4,6 +4,7 @@ import argparse
 import logging
 import os
 import queue
+import signal
 import sys
 import threading
 import wave
@@ -300,6 +301,13 @@ class NRSC5CLI:
                              data_service.mime_type)
 
 
+def signal_handler(sig, frame):
+    print("Caught signal")
+    os._exit(0)
+
+
 if __name__ == "__main__":
+    signal.signal(signal.SIGINT, signal_handler)
+
     cli = NRSC5CLI()
     cli.run()
```

Note that I had to use `os._exit()` instead of the usual `sys.exit()` because `sys.exit()` works by generating a `SystemExit` exception, which can occur inside the callback, in which case ctypes will ignore it.

`os._exit()` causes an immediate exit without any cleanup, so it might not necessarily be the best approach. The signal handler could perhaps instead inform `NRSC5CLI` that it should shut down.

argilo, 2025-02-25:
Here's an implementation of the "clean" shutdown approach, and it seems to work both for reading files and live playback with an RTL-SDR:

```diff
diff --git a/support/cli.py b/support/cli.py
index d6a177e..0c8c46d 100755
--- a/support/cli.py
+++ b/support/cli.py
@@ -4,6 +4,7 @@ import argparse
 import logging
 import os
 import queue
+import signal
 import sys
 import threading
 import wave
@@ -20,12 +21,20 @@ class NRSC5CLI:
         self.parse_args()
         self.audio_queue = queue.Queue(maxsize=64)
         self.device_condition = threading.Condition()
+        self.interrupted = False
         self.iq_output = None
         self.wav_output = None
         self.raw_output = None
         self.hdc_output = None
         self.audio_packets = 0
         self.audio_bytes = 0
+        signal.signal(signal.SIGINT, self._signal_handler)
+
+    def _signal_handler(self, sig, frame):
+        logging.info("Stopping...")
+        self.interrupted = True
+        with self.device_condition:
+            self.device_condition.notify()
 
     def parse_args(self):
         parser = argparse.ArgumentParser(description="Receive NRSC-5 signals.")
@@ -112,7 +121,7 @@ class NRSC5CLI:
 
         try:
             if self.args.r:
-                while True:
+                while not self.interrupted:
                     data = iq_input.read(32768)
                     if not data:
                         break
@@ -123,8 +132,6 @@ class NRSC5CLI:
             else:
                 with self.device_condition:
                     self.device_condition.wait()
-        except KeyboardInterrupt:
-            logging.info("Stopping...")
         except nrsc5.NRSC5Error as err:
             logging.error(err)
 
```

PR #365 (2025-02-25): Handle Ctrl+C better in the Python CLI
Fixes #361.

See that issue for a description of the problem and the approaches I tried to solve it.