theori-io/nrsc5 #373: Document the API

pclov3r, 2025-03-07:
Perhaps it might be helpful for others to make a documentation of the API.

argilo, 2025-03-07:
The C version is documented. As noted in the README, you can add `-DBUILD_DOC=ON` to your cmake command, and a `doc_doxygen/html` folder will appear in your build folder. Open `index.html` to see the docs.

pclov3r, 2025-03-07:
I guess i missed that my  bad :)

I'll close this now.

argilo, 2025-03-07:
It might be nice to have GitHub Actions generate the docs whenever the master branch is pushed, and upload the output to GitHub Pages.

argilo, 2025-03-07:
The Python API is not explicitly documented, but it does follow the C API closely, so the C documentation should be of some use.

pclov3r, 2025-03-07:
> The Python API is not explicitly documented, but it does follow the C API closely, so the C documentation should be of some use.

I'll re-open it because of that.

argilo, 2025-03-07:
If you'd like to take a crack at it, you'd just need to add docstrings to [support/nrsc5.py](https://github.com/theori-io/nrsc5/blob/master/support/nrsc5.py). Pydoc could then generate documentation in various formats.

Much of the information could be copied or adapted from [include/nrsc5.h](https://github.com/theori-io/nrsc5/blob/master/include/nrsc5.h).

argilo, 2025-05-04:
By the way, the C API documentation is now auto-generated by GitHub Actions and published here:

https://theori-io.github.io/nrsc5/c-api/