theori-io/nrsc5 #390: Windows Build

Birg55, 2025-04-07:
This may be a problem of my own making (I'm new at this), but when following the steps in the Readme for building for Windows when I get to the Build step, MSYS2 returns this result:
 
Ate the command liken prompt:

MSYS ~
$ nrsc5/support/msys2-build
:: Starting core system upgrade...
 there is nothing to do
:: Starting full system upgrade...
 there is nothing to do
**pacman: unknown option -- a**

The last line has me confused.

argilo, 2025-04-07:
Did you launch the "MinGW" version of the MSYS2 prompt?

Birg55, 2025-04-07:
I did not see in the readme a step for launching MinGW, so no.  I launched msys2.exe

I apologize, I am not a Coder

________________________________
From: Clayton Smith
Sent: Monday, April 7, 2025 5:57 PM
To: theori-io/nrsc5
Cc: Birg55; Author
Subject: Re: [theori-io/nrsc5] Windows Build (Issue #390)

[argilo]argilo left a comment (theori-io/nrsc5#390)<https://github.com/theori-io/nrsc5/issues/390#issuecomment-2784721163>

Did you launch the "MinGW" version of the MSYS2 prompt?

argilo, 2025-04-07:
It is mentioned in the readme:

> Open a terminal using the "MSYS2 MinGW 64-bit" shortcut.

I presume that doing that will resolve your issue, so I'll close this off for now. Let me know if it doesn't work.

closed 2025-04-07 by argilo

Birg55, 2025-04-08:
Sorry to bother again.  I was able to get nrsc5 to compile and play the sample 0 file, but I can't get it to do anything else.

The readme indicates:

"Once everything is built, you can run nrsc5 independently of MSYS2. Copy the following files from your MSYS2 mingw64 (or mingw32) directory (e.g. C:\msys64\mingw64\bin):

  *   libnrsc5.dll
  *
nrsc5.exe"

But the Readme doesn't suggest where I can copy the 2 files to.

Double clicking on nrsc5.exe causes a window to come up for a fraction of a second, and it closes.

If I try to run msys2.exe in its installed directory and use a command line example such as

$ nrsc5 100.6 0

I get:

-bash: nrsc5: command not found

If I try the example after changing directory:

$ cd ~/nrsc5/support

And execute:

$ nrsc5.exe -r sample 0

I get:

-bash: nrsc5.exe: command not found

Where can I move nrsc5.exe and libnrsc5.dll to?

And which shell program should I use to get nrsc5 to execute commands?

argilo, 2025-04-08:
nrsc5.exe is a command-line program, and so you'll need to execute it from a command window. (You can run "cmd.exe" to bring up the standard command window.)

It doesn't matter what directory you put `libnrsc5.dll` and `nrsc5.exe` in, but if that directory is not in your `PATH` then you'll need to change to that directory in the command window (using the `cd` command) before you run nrsc5.