theori-io/nrsc5 #391: Cross-Compile on macOS 15

markjfine, 2025-04-08:
Tested the updated cross-compile script for 64-bit Windows, which stopped while linking rtlsdr with several unreferenced calls to libusb.

Looking at the rtlsdr cmake trace, it found the version of libusb that was already installed by homebrew (1.0.28), not the desired one built under libusb-prefix (1.0.27).

This can be fixed by adding the path to libusb-prefix's pkgconfig folder to the beginning of the PKG_CONFIG_PATH and re-running the script from scratch.

Not a critical issue, but it does illustrate a potential build problem using the requisite downloaded modules (faad2, fftw, libao, libusb, or rtlsdr) if it isn't seen by pkgconfig first.

Incidentally, the cross-compiled version of nrsc5 did work well on Win11 once built and transferred to that box (typical wrestling with usb driver updating via Zadig, not withstanding).

argilo, 2025-04-08:
Is this a pre-existing problem, or did the recent changes in nrsc5 trigger it?

markjfine, 2025-04-08:
Pretty sure it was a pre-existing problem. I don't see how changes to the calling script could affect this.

markjfine, 2025-04-08:
Think I found the problem in `./rtlsdr-prefix/src/rtlsdr_external/src/CMakeLists.txt`:
```
########################################################################
# Find build dependencies
########################################################################
find_package(Threads)
find_package(PkgConfig)

if(PKG_CONFIG_FOUND)
    pkg_check_modules(LIBUSB libusb-1.0 IMPORTED_TARGET)
    set(LIBUSB_LIBRARIES "${LIBUSB_LINK_LIBRARIES}")
else()
    set(LIBUSB_LIBRARIES "" CACHE STRING "manual libusb path")
    set(LIBUSB_INCLUDE_DIRS "" CACHE STRING "manual libusb includepath")
endif()
```
As I assumed from the cmake trace, rtlsdr is ignoring what we're telling it to depend on, in favor of whatever pkgconfig finds first. May have to alter the PKG_CONFIG_PATH environment variable within nrsc5's cmake script prior to building rtlsdr, so it sees ${LIBUSB_PREFIX}/lib/pkgconfig first, but only for instances where USE_SYSTEM_LIBUSB is false.

markjfine, 2025-04-08:
Attempteed to change environment variable from within top level CMakeLists.txt however that only stays persistent for that instance of cmake. The ensuing make doesn't see it and therefore defaults back to using the installed libusb.

May make more sense to export a revised PKG_CONFIG_PATH from within the calling support script instead:
`export PKG_CONFIG_PATH=${prefix}/libusb_prefix/lib/pkconfig:$PKG_CONFIG_PATH`
prior to make.

markjfine, 2025-04-08:
Build works when export line is added to the calling script.
Tested resulting nrsc5 build on Win11 and it works without issue as well.

argilo, 2025-04-08:
Nice! Feel free to open a pull request if you're happy with that solution.

markjfine, 2025-04-08:
Am trying to as we type.

closed 2025-04-08 by argilo

markjfine, 2025-04-08:
Removed the 15 from the title. After investigation, it was clear that this was more specific to a user's configuration than the version of macOS.