theori-io/nrsc5 #398: Provide programs via alternative way

TheDaChicken, 2025-05-03:
Some HD stations like KNKX or WETS don't provide their multichannel programs in sis. 

This results in the client not being able to identify any programs unless audio packets or hdc packets were checked.

TheDaChicken, 2025-05-03:
My simple idea is that every PDU that is received, it can maybe send info about that audio program through the api. For example: #374

argilo, 2025-05-03:
I had a look over my collection of recordings, and noticed the same. There are a handful of stations (including some with multiple audio programs) that don't send any Audio Service Descriptors in SIS. So ASDs can't be reliably used as an indication of which audio programs exist.

The program number and PTY are also conveyed in the Header Expansion Field (HEF) of each PDU, and the HEF is mandatory when there are multiple audio programs. At present, nrsc5 decodes the entire HEF but uses only the program number. Perhaps the API should expose the contents of the HEF.

https://github.com/theori-io/nrsc5/blob/d0d91d01b0005a5da8330430df74c47772a8c673/src/frame.c#L48-L57

argilo, 2025-05-03:
We should probably also update the API documentation to indicate that not all stations broadcast Audio Service Descriptors.

markjfine, 2025-05-03:
Lol... So of course I use those SIS descriptors to update the program tabs and clickable info labels in `dui`.

argilo, 2025-05-03:
Assuming we expose PDU header information through the API, would you want it to work like SIS events and only be reported when it changes? And would you want all the audio programs reported in a single API event (like the `sis.audio_services` list), or one event per audio program?

TheDaChicken, 2025-05-03:
From the perspective that it might add more logging to console on ctl application, maybe every single it updated. A list would reduce the amount of calls into the python ctl for example.

markjfine, 2025-05-04:
I get the stream name and type from SIG, so that's not a problem.

From SIS, I get the audio program info from an "AudioComponent" line, or the data service info from a "DataComponent" line.

I think to keep things seamless, would it make sense to try to have the PDU reports emulate as if SIS does now.

Almost hate to ask, but did anyone notice if any data services were likewise omitted the same as audio programs?

argilo, 2025-05-04:
> I get the stream name and type from SIG, so that's not a problem.

SIG also appears to be optional, at least in cases where there's only a single audio program.

> Maybe to save on complexity (and redundancy), only provide PDU if SIS doesn't exist?

An audio program cannot be decoded without first decoding its PDU header, so we should probably take the header to be the authoritative source for the *existence* of an audio program, with additional metadata optionally provided by SIS and/or SIG.

It's definitely a pain that audio program information comes from three sources.

> Almost hate to ask, but did anyone notice if any data services were likewise omitted the same as audio programs?

I haven't looked into that just yet. But one weird situation I *have* seen is that some stations send AAS files without a corresponding entry in the SIG table. They appear to be sending station logos for defunct subchannels.

markjfine, 2025-05-04:
I suppose it makes more sense to prioritise PDU reporting over the other two then.

Only other concern is how often the PDU reporting occurs, and how much extra processing this adds to nrsc5 itself, as well as any application that uses it. I remember that when I originally added the extra information to nrsc5-gui, python had issues dealing with the audio packets on a machine with moderate processing power (a 2012-ish MBP running Catalina). That's one of the reasons I went back to a spawned nrsc5 process that just reacted to a stdout pipe in nrsc5-dui. Was more important that the audio output took priority over display processing.

argilo, 2025-05-04:
> Only other concern is how often the PDU reporting occurs, and how much extra processing this adds to nrsc5 itself, as well as any application that uses it.

The reporting would be once at the start of a stream, so the burden on the application would be near zero. The extra processing time would also be minimal.

> That's one of the reasons I went back to a spawned nrsc5 process that just reacted to a stdout pipe in nrsc5-dui. Was more important that the audio output took priority over display processing.

I'm surprised that was necessary. On my system, the nrsc5 executable uses about 5% CPU, compared to 6% for cli.py. The C code is doing all the heavy lifting, and the Python API is just a thin wrapper.

It could be you were running into trouble because of the enormous audio bursts (which will be fixed by #399), which may have resulted in blocking while waiting for audio output to complete.

Do you think you might eventually switch over to the API? If there's something that makes it impractical, feel free to open an issue for it.

markjfine, 2025-05-04:
> Do you think you might eventually switch over to the API? If there's something that makes it impractical, feel free to open an issue for it.

Probably won't be doing much more with it other than minor improvements, as critical/necessary. There's a long story of how I originally started updating nrsc5-gui as something to kill lots of down time while being stuck up in NJ several years back. I'm kinda at the stage where I'd rather help others do better things, low key.

argilo, 2025-05-04:
OK, that's fine. I just want to make sure we know about any obstacles to API use (e.g. #372, #373, #397) so we can get them out of the way.

PR #402 (2025-05-07): Report information from audio PDU headers
Fixes #374.
Fixes #398.

Neither SIS audio descriptors nor SIG are mandatory, leaving applications with no reliable way to know of the existence of an audio program. Audio PDU headers are always present, and convey useful information about audio programs, so I think it makes sense to expose this information through the API.

Here I've added a new event (`NRSC5_EVENT_AUDIO_SERVICE`) which exposes ~a linked list of~ audio services. The following information is available:

* program number
* access (public / restricted)
* program type
* codec mode
* blend [... cut]